embox/embox #721: New README and contributing guides

abusalimov, 2015-04-14:
The current README file is completely outdated.

- [ ] Provide more actual project description
- [ ] Add [contributing guide-lines](https://help.github.com/articles/setting-guidelines-for-repository-contributors/) (CONTRIBUTING.md file)
- [ ] Quick start

sikmir, 2015-05-24:
[Quick start] What about advising to use qemu-linaro for ARM emulation?

AntonKozlov, 2015-05-25:
qemu-linaro have beagleboard emulation, which is our only available arm/qemu platform

sikmir, 2015-05-25:
@AntonKozlov, I just suggest to mention about it in Quick Start guide.

abusalimov, 2015-05-25:
@sikmir Could you suggest a PR for that, please? The current setup works for me just fine, likely because I have an environment similar to that on Travis, so that I'm afraid I won't be able to reproduce the issue and properly fix it.

sikmir, 2015-05-25:
@abusalimov It's not an issue in fact, just a note for archlinux users.
Case 1:
```fish
~ sudo pacman -S qemu
~ qemu-system-arm --machine help | grep overo
(nothing)
```
Case 2:
```fish
~ yaourt qemu-linaro
~ qemu-system-arm --machine help | grep overo
overo                Gumstix Overo board (OMAP3530)
```
I don't know why follow one works for travis:
```fish
sudo apt-get install -qq qemu qemu-system
```

AntonKozlov, 2015-05-26:
travis seems to use some ubuntu-based distro (may be ubuntu itself). It's qemu built with linaro patches.

https://launchpad.net/qemu-linaro

`Want native binaries? The qemu-system package in Ubuntu Maverick and later already includes some Linaro changes ...`

Deryugin, 2015-05-26:
Debian has the same problem with qemu from default repos.
Also, build fails if */bin/sh* points to */bin/dash* (*"Bad substitution"*), and it's default for some debian and ubuntu versions.
Issue could be fixed by linking */bin/sh* to */bin/bash*.

PR #1083 (2017-06-17): contributing: Add initial version of the guide
Better late than never.

Closes #721.
closed 2017-06-17 by abusalimov