embox/embox #728: Xen build failed

AleksandraButrova, 2015-04-16:
```
EMBOX_GCC_LINK=full ./mk/extbld/arch-embox-gcc ./mk/extbld/toolchain_test.c -o ./build/base/obj/toolchain_test
./build/base/obj/embox.o: In function `rootfs_mount':
src/fs/rootfs.c:31: undefined reference to `mount'
./build/base/obj/embox.o: In function `serial_close':
src/drivers/tty/serial/idesc_serial.c:192: undefined reference to `file_desc_destroy'
collect2: error: ld returned 1 exit status
mk/extbld/toolchain.mk:72: recipe for target 'build/base/obj/toolchain_test' failed
make[4]: *** [build/base/obj/toolchain_test] Error 1
mk/build.mk:20: recipe for target 'build' failed
make[3]: *** [build] Error 2
mk/load.mk:42: recipe for target 'build' failed
make[2]: *** [build] Error 2
mk/main.mk:32: recipe for target 'build' failed
make[1]: *** [build] Error 2
Makefile:38: recipe for target 'all' failed
make: *** [all] Error 2
```

Deryugin, 2015-04-17:
Can't reprocude this issue. @AleksandraButrova, can you try running "make clean && make"?

AntonKozlov, 2015-04-17:
Hmm, me too. 
I forgot boot.S file. When I commit it, xen/debug successfully built

closed 2015-04-19 by AntonKozlov