embox/embox #760: TASK_RESOURCE_DECLARE: use or drop

AntonKozlov, 2015-05-29:
There is handy TASK_RESOURCE_DECLARE macro, that allow to skip offset managment, required by TASK_RESOURCE_DEF

Compare 
https://github.com/embox/embox/tree/master/src/kernel/task/resource/u_area/u_area.c
and
https://github.com/embox/embox/tree/master/src/compat/posix/fs/umask.c

This scheme isn't best, but since we adopted it, it's better to keep it consistent

Deryugin, 2015-05-29:
This way you need to leave trailing comma in TASK_RESOURCE_DECLARE. Is there any macro magic to avoid it?

AntonKozlov, 2015-05-29:
first comma if resource is public?
Another macro could be solution.
But OK, issue confirmed

Deryugin, 2015-05-29:
Actually I meant comma after VA_ARGS. It's not required if VA_ARGS are empty, so you can't simple add comma to macro declaration

>TASK_RESOURCE_DECLARE(static,
>task_resource_umask,
>mode_t,
>.init = task_umask_init,
>.inherit = task_umask_inherit**,**
>);
Look at the 5-th line.

But first comma should be a problem as well.

AntonKozlov, 2015-05-29:
OK, will do something with it

abusalimov, 2015-05-29:
Changing the macro [definition](https://github.com/embox/embox/blob/e3739c3e589b59cf551f58ff516a2e55de41476f/src/include/kernel/task/resource.h#L38-L45) from
```c
#define TASK_RESOURCE_DECLARE(desc_modifies, desc_name, resource_type, ...) \
	static size_t __ ## desc_name ## _offset; \
	desc_modifies const struct task_resource_desc desc_name = { \
		__VA_ARGS__ \
		.resource_size = sizeof(resource_type), \
		.resource_offset = &__ ## desc_name ## _offset, \
	}; \
	TASK_RESOURCE_DEF(desc_name, resource_type)
```

to
```c
#define TASK_RESOURCE_DECLARE(desc_modifies, desc_name, resource_type, ...) \
	static size_t __ ## desc_name ## _offset; \
	desc_modifies const struct task_resource_desc desc_name = { \
		.resource_size = sizeof(resource_type), \
		.resource_offset = &__ ## desc_name ## _offset, \
		## __VA_ARGS__ \
	}; \
	TASK_RESOURCE_DEF(desc_name, resource_type)
```

should do the trick.

AntonKozlov, 2015-05-29:
yep, agree

Deryugin, 2015-05-29:
And what about leading comma in case of non-present modifier?

AntonKozlov, 2015-05-29:
Something like?

#define TASK_STATIC_RESOURCE(a, b, ...) \
    TASK_RESOURCE_DECLARE(static, a, b, ##VA_ARGS)

#define TASK_RESOURCE(a, b, ...) \
    TASK_RESOURCE_DECLARE(, a, b, ##VA_ARGS)

I'll do later, if all of you agree

Deryugin, 2015-05-29:
It's ok IMO

closed 2020-01-27 by anton-bondarev