embox/embox #813: Support building under Mac OS X

onegray, 2015-08-14:
At the moment `make` fails while building `openlibm`:

    .../arch-embox-gcc -include ...   -I ...  -D__ELF__  -fno-gnu89-inline -std=c99 -Wall -O3  -DASSEMBLER -D__BSD_VISIBLE -Wno-implicit-function-declaration -m32 -march=i386 -c src/s_isinf.c -o src/s_isinf.c.o
    <inline asm>:1:1: error: unknown directive
    .weak isinff
    ^
    1 error generated.
    make[6]: *** [src/s_isinf.c.o] Error 1

AntonKozlov, 2015-08-14:
Could you provide name of template you're trying to build?

onegray, 2015-08-14:
make confload-x86/qemu
    make

AntonKozlov, 2015-08-17:
You're right, build failed on Mac OS.
I've made couple of fixes in buildsystem #815 and updated title page with info on crosstool #817. Could you check that the last is clear? In general, crosscompiler should be build when you're on MacOS (our `crosstool` script suggested). Links are in that PR. Please let us know if there are any issues with crosstool or something unclear. 
Thanks!

onegray, 2015-08-20:
Finally, using `crosstool` I was able to successfully build it under Mac OS X 10.10.3.
Thanks!

closed 2015-08-20 by onegray