embox/embox #930: sudo ./scripts/qemu/auto_qemu fails

AnastasiaSulyagina, 2016-04-19:
Fails on OSX 10.11.4.

Log:
AUTOQEMU_ARCH: x86 (guessed)
AUTOQEMU_MEM: 256 (guessed)
AUTOQEMU_NICS: (guessed)
AUTOQEMU_NICS_CONFIG: (got)
egrep: /proc/cpuinfo: No such file or directory
VT is not supported by CPU
./scripts/qemu/auto_qemu: line 85: lsmod: command not found
no kvm kernel module loaded
AUTOQEMU_KVM_ARG: (guessed)
AUTOQEMU_LOAD_ARG: -kernel ./build/base/bin/embox (guessed)
AUTOQEMU_NOGRAPHIC_ARG: -nographic (guessed)
AUTOQEMU_MACHINE: (guessed)
./scripts/qemu/auto_qemu: line 143: declare: -A: invalid option
declare: usage: declare [-afFirtx] [-p] [name[=value] ...]
printf: QEMU: expected numeric value
printf: emulator: expected numeric value
printf: version: expected numeric value
printf: 0,: not completely converted
printf: Copyright: expected numeric value
printf: (c): expected numeric value
printf: 2003-2008: not completely converted
printf: Fabrice: expected numeric value
printf: Bellard: expected numeric value
./scripts/qemu/auto_qemu: line 157: declare: -A: invalid option
declare: usage: declare [-afFirtx] [-p] [name[=value] ...]
qemu-system-i386 -kernel ./build/base/bin/embox -m 256 -nographic

Embox kernel start
	unit: initializing embox.mem.vmem_alloc: done
	unit: initializing embox.kernel.task.task_resource: done
	unit: initializing embox.mem.phymem: start=0x09af6000, end=0x10100000, size=106995712
done
runlevel: init level is 0
	unit: initializing embox.profiler.tracing: done
	unit: initializing embox.driver.interrupt.i8259: done
	unit: initializing embox.driver.clock.pit: done
	unit: initializing embox.kernel.time.jiffies: done
	unit: initializing embox.kernel.time.kernel_time: done
	unit: initializing embox.kernel.task.task_table: done
	unit: initializing embox.kernel.task.kernel_task: done
	unit: initializing embox.kernel.sched.sched: done
	unit: initializing embox.fs.driver.fs_driver: done
	unit: initializing embox.kernel.time.timekeeper: done
	unit: initializing embox.fs.node: done
	unit: initializing embox.security.smac: done
	unit: initializing embox.security.seculog_file: done
	unit: initializing embox.mem.static_heap: done
	unit: initializing embox.kernel.time.timer_handler: done
	unit: initializing embox.fs.buffer_cache: done
	test: running embox.test.framework.mod.member.ops_test . done
	test: running embox.test.kernel.timer_test ..... done
	test: running embox.test.recursion . done
	test: running embox.test.posix.sleep_test ...... done
	test: running embox.test.stdio.printf_test ......... done
	test: running embox.test.posix.getopt_test .............. done
	test: running embox.test.posix.poll_test ......... done
	test: running embox.test.posix.select_test ...... done
	test: running embox.test.posix.pipe_test ... done
	test: running embox.test.posix.ppty_test ... done
	test: running embox.test.stdlib.bsearch_test ...... done
	test: running embox.test.stdlib.qsort_test ....... done
	test: running embox.test.posix.environ_test ......... done
	test: running embox.test.util.array_test . done
	test: running embox.test.util.bit_test ..... done
	test: running embox.test.util.slist_test .................. done
	test: running embox.test.util.tree_test ........... done
	test: running embox.test.util.indexator_test ........... done
	test: running embox.test.math.math_test .. done
	test: running embox.test.mem.pool_test ...... done
	test: running embox.test.util.hashtable_test ... done
	unit: initializing embox.driver.ide: done
	test: running embox.test.critical .. done
runlevel: init level is 1
	unit: initializing embox.net.neighbour: done
	unit: initializing embox.driver.block: done
	unit: initializing embox.net.tcp: done
	unit: initializing embox.driver.pci: done
	unit: initializing embox.net.net_entry: done
	unit: initializing embox.driver.net.loopback: done
	unit: initializing embox.fs.rootfs: initfs_mount: unpack initinitfs at 0x001f04a4 into /
done
	test: running embox.test.mmu_core .. done
	test: running embox.test.net.socket_test ............................ done
	test: running embox.test.net.inet_socket_test ...... done
	test: running embox.test.net.inet_dgram_socket_test ............. done
	test: running embox.test.net.inet_stream_socket_test .............. done
	test: running embox.test.net.raw_socket_test . done
	pci: virtio driver inserted
	unit: initializing embox.driver.serial.i8250: done
	unit: initializing embox.kernel.sched.sched_ticker_preempt: done
	unit: initializing embox.fs.driver.tmpfs: done
	unit: initializing embox.fs.driver.ramfs: done
runlevel: init level is 2
	unit: initializing embox.init.start_script: 
Started shell [tish] on device [ttyS0]
loading start script:
> ifconfig lo 127.0.0.1 netmask 255.0.0.0 up 
> route add 127.0.0.0 netmask 255.0.0.0 lo 
> ifconfig eth0 10.0.2.16 netmask 255.255.255.0 hw ether AA:BB:CC:DD:EE:02 up 
eth0: unknown interface
ifconfig: Command returned with code -19: No such device
error: Unknown error code. PLEASE REPORT ISSUE 519!
Failed to get into level 3, current level 2

anton-bondarev, 2016-04-20:
Thanks for the report!

Your command for qemu start is
 
> qemu-system-i386 -kernel ./build/base/bin/embox -m 256 -nographic

It means that there isn't any network interface in your qemu.
Did you start **auto_qemu** script with **sudo**?
If you don't need network now you can just remove its configuration commands from ./conf/start_script.inc

AnastasiaSulyagina, 2016-04-20:
@anton-bondarev , fails with 
>> qemu-system-i386 -kernel ./build/base/bin/embox -m 256 -nographic
as well. Log is the same.
Yes, started both with sudo and I do need network.

anton-bondarev, 2016-04-20:
@AnastasiaSulyagina 
Please try this command line

> sudo qemu-system-i386 -kernel ./build/base/bin/embox -m 256 -net nic,vlan=0,model=virtio,macaddr=AA:BB:CC:DD:EE:02 -net tap,script=./scripts/qemu/start_script,downscript=./scripts/qemu/stop_script,vlan=0 -nographic

Our script (auto_qemu) why that didn't recognize net-card from the configuration.
Correct message have to  

> AUTOQEMU_ARCH: x86 (guessed)
> AUTOQEMU_MEM: 256 (guessed)
> AUTOQEMU_NICS: virtio (guessed)
> AUTOQEMU_NICS_CONFIG: (got)

AnastasiaSulyagina, 2016-04-20:
@anton-bondarev 
>> qemu-system-i386: -net tap,script=./scripts/qemu/start_script,downscript=./scripts/qemu/stop_script,vlan=0: could not open /dev/tap0: No such file or directory

anton-bondarev, 2016-04-20:
@AnastasiaSulyagina 
Hm, I think it's the problem not in QEMU.
there is no 'tap' device support in macos. We try to suggest a way to resolve this problem soon.

AnastasiaSulyagina, 2016-04-20:
@anton-bondarev , thanks.

AntonKozlov, 2016-04-20:
Hi, All,

Agree, seems like tap networking will not work out-of-box on macos (darwin). There is some info about proper qemu networking with tap on http://drupal.bitfunnel.net/drupal/macosx-bridge-qemu, hope it works. In case of success we'll modify our qemu scripts.

alexkalmuk, 2016-04-20:
Hi guys,
What if tuntap driver was not pre-installed at all?
May be it would be suitable just like a workaround to setup the TUNTAP tool for OSX at first - http://tuntaposx.sourceforge.net/index.xhtml (or just do "brew install tuntap" for another one)
I am in doubt that it resolves all the world's problems, but most likely it creates /dev/tapX.

AnastasiaSulyagina, 2016-04-20:
Thanks, @alexkalmuk. Worked for me.

Deryugin, 2016-05-06:
I guess we can close this issue as soon as we add OSX to "Preparing environment" in README.md

AntonKozlov, 2016-05-27:
Since #939 docker is the official way to develop on MacOSX. 
Feel free to reopen in case of issues.

closed 2016-05-27 by AntonKozlov