embox/embox #947: Embox + docker on mac os X

AnastasiaSulyagina, 2016-06-05:
Не могу запустить ембокс в докере. С командой 
```
sudo qemu-system-i386 -kernel ./build/base/bin/embox -m 256 -net nic,vlan=0,model=virtio,macaddr=AA:BB:CC:DD:EE:02 -net tap,script=./scripts/qemu/start_script,downscript=./scripts/qemu/stop_script,vlan=0 -nographic
```
все нормально, а с 
```
dr ./scripts/qemu/auto_qemu -s -S
```
висит с 
```
ioctl(TUNSETIFF): Device or resource busy
Enable IP Forwarding for eth0
net.ipv4.ip_forward = 1
```

AnastasiaSulyagina, 2016-06-05:
И как раньше билдила через crosstool теперь не могу для дебага запустить

AntonKozlov, 2016-06-06:
Можно и без докера

-s -S останавливает qemu до подсоединения дебагера. Пока дебагером не подцепиться, дальше загрузка не пойдет.

Насколько понимаю, поведение одинаковое с докерм и без?

AnastasiaSulyagina, 2016-06-06:
Дебаггер подцеплен, но он не слышит без докера. А как дебаггер к докеру подключить не совсем понятно.

AntonKozlov, 2016-06-07:
Последовательность отладки
1. запустить qemu -s -S
2. подцепиться дебагером (remote в eclipse, `target extended-remote :1234` в gdb)
3. Сказать continue в дебагере или eclipse

Не совсем понял, что значит "не слышит без докера"

Для отладки в докере вместо `gdb` на хост машине надо запускать gdbhostwrapper (https://github.com/embox/embox/wiki/Emdocker#gdb). Для эклипса есть готовая конфигурация для дебага scripts/docker/qemu_on_docker.launch, она должна появиться в меню дебаг конфигураций.

Если всё равно не сработает, дайте пожалуйста последовательность, что Вы делали и что видели по пути.

AnastasiaSulyagina, 2016-06-07:
Я сейчас пока поставила виртуалку убунты, там доделаю, а потом когда время будет, попробую повторить. 
Не слышит - не подключается и выдает timeout.

anton-bondarev, 2020-01-24:
@AntonKozlov Won't fix?

closed 2020-01-24 by anton-bondarev