embox/embox #949: mips/qemu: soft/hard-float ABI inconsistency

frantony, 2016-06-15:
Here is a fragment from mips/qemu build log:

````
mips-elf-ld: Warning: build/base/obj/embox-2.o uses -msoft-float
  (set by ./build/base/obj/module/embox/arch/mips/kernel/interrupt.o),
  ./build/base/obj/module/embox/arch/mips/libarch.a(setjmp.o) uses -mhard-float
````

anton-bondarev, 2016-08-19:
@frantony I couldn't reproduce this warning.
I have 

> mips-elf-gcc --version
mips-elf-gcc (GCC) 4.9.1
Copyright (C) 2014 Free Software Foundation, Inc.

and 

> uname -a
Linux anton-Inspiron-1520 3.13.0-24-generic #47-Ubuntu SMP Fri May 2 23:30:00 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux

frantony, 2016-08-20:
> I have
>> mips-elf-gcc --version
>> mips-elf-gcc (GCC) 4.9.1

You use gcc-4.9.1; please try gcc-4.9.2 from recent [embox crosstool](https://github.com/embox/crosstool).

frantony, 2017-03-03:
ping

anton-bondarev, 2017-03-06:
pong:)

We couldn't represented this problem before, but now we'll try to use mips-linux-gcc from debian packets. I hope we'll represent it.

AntonKozlov, 2017-03-08:
Hi,

I can see same warning on https://github.com/embox/crosstool/releases/download/current/mips-elf-toolchain.tar.bz2

Seems like gcc bug that was fixed in newer versions of gcc
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=64569

One solution is to bump gcc version, another is to workaround a bug like linux did 
https://patchwork.linux-mips.org//patch/8355/
Since it's just single config change, I suppose this. Will push PR shortly

Another question is how other architectures affected? gcc bug description talking about mips enforced checks, does this mean that others are just silently ignore `-msoft-float`?

AntonKozlov, 2017-03-09:
Just note, I renamed that crosstool release to https://github.com/embox/crosstool/releases/tag/2.25.1-4.9.2-7.9

AntonKozlov, 2017-03-13:
@frantony Thanks! The fix is merged in, if it was not sufficient, please reopen

PR #1050 (2017-03-13): Workaround mips gcc bug of soft/hard float mixed
Fixes #949
closed 2017-03-13 by AntonKozlov