embox/embox #954: x86/user_apps broken, not builds

AntonKozlov, 2016-06-28:
Seems that this template haven't built for a long time.
There are some problems with stat/new_stat that I can't resolve in short time.
```
gcc  -pipe --debug-prefix-map=`pwd`= --debug-prefix-map=./= -fno-strict-aliasing -fno-common -Wall -Werror -Wundef -Wno-trigraphs -Wno-char-subscripts  -Wno-nonnull-compare -Wformat -std=gnu99 -O0 -g -nostdinc -march=i386 -m32 -fno-stack-protector -Wno-array-bounds -U__linux__ -Ulinux -U__linux -D__EMBOX__ -D__unix -imacros build/base/gen/config.lds.h -Ibuild/base/include -I./src/include -I./src/arch/x86/include -Ibuild/base/gen/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/arch/x86/include  -I./src/compat/linux/include -I./src/compat/posix/include -I./src/compat/libc/include -nostdinc -MMD -MP  -include build/base/gen/include/config/embox/kernel/syscall.h -D__EMBUILD_MOD__=embox__kernel__syscall -c -o build/base/obj/src/kernel/syscall/syscall.o src/kernel/syscall/syscall.c
In file included from ./src/compat/posix/include/unistd.h:17:0,
                 from src/kernel/syscall/syscall.c:10:
./src/compat/posix/include/sys/stat.h:100:25: error: expected ':', ',', ';', '}' or '__attribute__' before '.' token
 #define st_atime st_atim.tv_sec      /* Backward compatibility */
                         ^
src/kernel/syscall/syscall.c:102:8: note: in expansion of macro 'st_atime'
   long st_atime;
        ^
src/kernel/syscall/syscall.c:114:1: error: size of unnamed array is negative
 static_assert(sizeof(struct new_stat) == 60);
 ^
src/kernel/syscall/syscall.c: In function 'sys_newfstat':
src/kernel/syscall/syscall.c:132:10: error: 'struct new_stat' has no member named 'st_blksize'
  new_stat.st_blksize = stat.st_blksize;
          ^
src/kernel/syscall/syscall.c:133:10: error: 'struct new_stat' has no member named 'st_blocks'
  new_stat.st_blocks = stat.st_blocks;
          ^
src/kernel/syscall/syscall.c:134:10: error: 'struct new_stat' has no member named 'st_atim'
  new_stat.st_atime = stat.st_atime;
          ^
src/kernel/syscall/syscall.c:135:10: error: 'struct new_stat' has no member named 'st_mtim'
  new_stat.st_mtime = stat.st_mtime;
          ^
src/kernel/syscall/syscall.c:136:10: error: 'struct new_stat' has no member named 'st_ctim'
  new_stat.st_ctime = stat.st_ctime;
          ^
src/kernel/syscall/syscall.c: At top level:
cc1: error: unrecognized command line option "-Wno-nonnull-compare" [-Werror]
cc1: all warnings being treated as errors
```

AntonKozlov, 2016-06-28:
Can't resolve stat issue, @anton-bondarev could you reassign it to some one?

anton-bondarev, 2016-06-28:
@AntonKozlov Yes, of cause, I can:) @Deryugin or me fix it! You work on the issue in [this branch](https://github.com/embox/embox/tree/954-fix-x86-user_apps)?
But I think we should also decrease number of templates.

AntonKozlov, 2016-06-28:
Yes, suppose you could make changes in that branch.
This template is valuable, since it should have mmu working

anton-bondarev, 2016-06-28:
@AntonKozlov I fix it, check please

anton-bondarev, 2016-06-28:
Closed by #955

closed 2016-06-28 by anton-bondarev