embox/embox #958: [Clang] ARRAY_SPREAD_DECLARE and double const

Kakadu, 2016-06-29:
Там шланг находит, что жопа с появлением двойной константности. Надо либо объявлять это везде без константности (а это много мест), либо стереть константность в `__ARRAY_SPREAD_DECLARE`. Прибить опцией командной строки пока не получается -- с виду опций вида -Wno-мой-самый-любимый-ворниг нету. Позже спрошу у буржуев как надо.

@anton-bondarev  @Deryugin 

````
./src/include/kernel/task/resource.h:65:1: error: duplicate 'const' declaration specifier [-Werror,-Wduplicate-decl-specifier]
ARRAY_SPREAD_DECLARE(const task_notifing_resource_hnd,
^
./src/include/util/array.h:205:2: note: expanded from macro 'ARRAY_SPREAD_DECLARE'
        __ARRAY_SPREAD_DECLARE(element_type, name)
        ^
./src/include/../../src/util/array_spread/array_impl.h:84:30: note: expanded from macro '__ARRAY_SPREAD_DECLARE'
        EXTERN_C element_t volatile const array_nm[]
```

abusalimov, 2016-06-29:
> Надо либо объявлять это везде без константности (а это много мест),

From [util/array.h](https://github.com/embox/embox/blob/b5d9e149b4e44a95bd291e47dae11bd69e85b94c/src/include/util/array.h):
<pre>
 *     Considering current use cases and some implementation issues, spread
 * arrays are always allocated in read-only data section as if you have defined
 * a regular array with @c const modifier (although you might not have to). To
 * prevent confusion and to take advantage of compiler type checking <b>always
 * include @c const modifier to element type</b> when defining spread array
 * with #ARRAY_SPREAD_DEF() and its derivatives or when declaring it using
 * @c extern.
</pre>

P.S. English, please. :beers:

Kakadu, 2016-06-29:
OK, @abusalimov , so which of three mentioned options do you recommend?

abusalimov, 2016-06-30:
Fixing the implementation seems to be the right way.
```diff
-        EXTERN_C element_t volatile const array_nm[]
+        EXTERN_C element_t volatile array_nm[]
```

HaK1R, 2016-07-02:
It states that the array is read only. By removing this, all arrays of
pointers become not const. So in this case you have to find all places like:
ARRAY_SPREAD_DEF(const mytype *)
and replace them with
ARRAY_SPREAD_DEF(const mytype * const)

That's why this qualifier was set in this macro (but yes, unfortunately it
produces a double const for non-pointer types).

Probably the other way is to hide one of the `const` by typedef'ing the
element_t.

abusalimov, 2016-07-02:
@HaK1R hmm, I didn't think about that, it seems you're right, thank you. The code in master seems to be broken now.

> Probably the other way is to hide one of the `const` by typedef'ing the element_t.
I guess one could also workaround that using `typeof`.

abusalimov, 2016-07-02:
@Kakadu 

> Прибить опцией командной строки пока не получается -- с виду опций вида -Wno-мой-самый-любимый-ворниг нету

`-Wno-duplicate-decl-specifier` seems to do the trick.

I would suggest to `git revert f93c42e` and also add missing `const` specifiers as suggested by @HaK1R, i.e.:
```diff
-ARRAY_SPREAD_DEF(const mytype *)
+ARRAY_SPREAD_DEF(const mytype * const)
```

anton-bondarev, 2016-08-09:
@HaK1R @abusalimov I replaced ARRAY_SPREAD_DEF() with pointer types as you described in PR #974 Should I do anything else?
About commit reverting, I think that I just make a new commit and it will be enough.

abusalimov, 2016-08-09:
I'll try to take a look this evening.

Regards,
Eldar

anton-bondarev, 2016-08-22:
Fix by #974

closed 2016-08-22 by anton-bondarev