embox/embox #997: Implement ferror and friends

alexkalmuk, 2016-10-06:
There are a bunch of functions (clearerr, feof, ferror, fileno) that are used to check and reset stream status.

MrSquanchee, 2020-03-03:
Hii @alexkalmuk I would like to work on this issue.

alexkalmuk, 2020-03-03:
Hi, well, you can get a try.

alexkalmuk, 2020-03-03:
I'd suggest you first to check whether these functions are not implemented yet.

alexkalmuk, 2020-03-03:
> And stubs for the other functions ferror() , feof() and clearerr() are defined here.

So it looks like we should implement these three functions.

MrSquanchee, 2020-03-04:
@alexkalmuk . I am a lil bit stuck here. I want to know whether the `stdio` file functions implement an error field in the `flags` variable of file_struct or not or the error resolution is solely done with `errno` variable. If not I would like to change the code of all the file IO functions a lil bit to accommodate for a file error field and the `EOF` field in the `flags` variable.
Am I thinking in the correct direction ??? Guidance Required.

MrSquanchee, 2020-03-04:
>  or the error resolution is solely done with errno variable.
As far as I have seen this seems to be the most obvious implementation.

alexkalmuk, 2020-03-04:
You can probably start with tests for these functions, and check how they work on Linux first. So after that, we will have a bunch of tests and will think about how to implements this in Embox.

MrSquanchee, 2020-03-05:
I propose following changes and want your advise ....
1. I want to add following lines in file_struct.h or stdio.h (I really don't know)
`#define _IO_EOF_SEEN          0x0010`
`#define _IO_ERR_SEEN          0x0020`
To check for the EOF for `feof()` and ERR for `ferror()` and later clear them from `file->flags` in `clearerr()`.

2. I want to add error handling in [fflush.c](https://github.com/embox/embox/blob/master/src/compat/libc/stdio/fflush.c) because right now fflush doesn't tell whether the [lib_ob_forceflush](https://github.com/embox/embox/blob/master/src/compat/libc/stdio/fflush.c#L19) worked or not.

3. I want to change the behavior of [fread](https://github.com/embox/embox/blob/master/src/compat/libc/stdio/fread.c) when encountering an error. That is, returning the error with setting the the `_IO_EOF_SEEN` and `_IO_ERR_SEEN` fields in `file->flags`.

4. I want to add Similar functionalities in [fseek](https://github.com/embox/embox/blob/master/src/compat/libc/stdio/fseek.c) and [fwrite](https://github.com/embox/embox/blob/master/src/compat/libc/stdio/fwrite.c).

5. As other file IO functions like `fgetc`, `fgets`, `fputc`, `fputs` depend on the above functions, they will also be covered by `ferror` and `feof`.

What do you say fellow developer ??
:)

alexkalmuk, 2020-03-05:
@MrSquanchee Let's start with feof and clearerr which clears only EOF flag :)

alexkalmuk, 2020-03-05:
As I understand `feof` should return whether EOF flag is already set. How it can be tested? ILet's consider a simple regular file with one symbol - `a`.

```
f = fopen("a.txt", "r");
test_assert_equal(feof(f). 0);
test_assert_not_equal(fgetc(f). EOF);
test_assert_equal(feof(f). 0);
test_assert_equal(fgetc(f). EOF);
test_assert_not_equal(feof(f). 0);
fclose(f);
```

What do you think?

alexkalmuk, 2020-03-05:
btw I think the changes you proposed may be good, but I would suggest to verify this only for `read`, without changing fflush, fseek, etc.

MrSquanchee, 2020-03-05:
> I would suggest to verify this only for `read`, without changing fflush, fseek, etc.

Roger that ..... :)

MrSquanchee, 2020-03-05:
I implemented all the functions and flags and put the flags in `file_struct.h`.
I don't know where to put this test file and how to run the test.
Would you describe the process a lil bit to me ??
I am a noobie when it comes to testing, so please bear with me.

PR #1782 (2020-03-09): Implemented functions feof(), ferror() and clearerr()
closes #997: Implementing ferror and friends and closes #1746
I have added functionality for detection of EOF and any ERR in the
files->flags by creating macros IO_EOF_ (0x0010) and IO_ERR_ (0x0020)
and updating files->flags, whenever any file IO function encounters
EOF or any kind of error occurs, which is detected by functions
feof() and ferror() and cleared by clearerr().
file IO functions where EOF may occur -> fread()
file IO functions where ERR is set    -> fflush(), fread(), fseek()
and fwrite().
commit 93ad5228 (2020-03-09): Implemented functions feof(), ferror() and clearerr()

Solving Issue #997: Implementing ferror and friends and Issue #1746
I have added functionality for detection of EOF and any ERR in the
files->flags by creating macros IO_EOF_ (0x0010) and IO_ERR_ (0x0020)
and updating files->flags, whenever any file IO function encounters
EOF or any kind of error occurs, which is detected by functions
feof() and ferror() and cleared by clearerr().
file IO functions where EOF may occur -> fread()
file IO functions where ERR is set    -> fread() and fwrite().

Implementing ferror and friends and closes #997  [... cut]
closed 2020-03-09 by alexkalmuk