embox/embox #1093: Build errors

sikmir, 2017-07-04:
Some errors with gcc 7.1.1:
```bash
src/tests/stdio/printf_test.c: Function ”__test_case_at_line_113”:
src/tests/stdio/printf_test.c:122:23: error: ”snprintf” output truncated before the last format character [-Werror=format-truncation=]
  test_assert_equal(4, snprintf(dest, 4, "1234"));
...
src/tests/stdio/printf_test.c:125:23: error: ”12345678” directive output truncated writing 8 bytes into a region of size 2 [-Werror=format-truncation=]
  test_assert_equal(8, snprintf(dest, 2, "12345678"));
```
Template: x86/qemu

AntonKozlov, 2017-07-05:
That strange, I have gcc (GCC) 7.1.1 20170528, on commit
```
$ git show
commit c2497da6ad356d8aef04d91bd313bc442330a194 (HEAD)
Merge: 3585661aa ffcbe9157
Author: Denis Deryugin <deryugin.denis@gmail.com>
Date:   Sat Jul 1 17:10:37 2017 +0300

    Merge pull request #1088 from anna-marina/gcc-7-warnings
    
    Fix GCC7-specific build errors
```
I have that warnings as warnings, not errors. Build completes successfully. Could you provide details about commit you've tested, extra CFLAGS in shell environment (like one with '-Wall'), etc?

sikmir, 2017-07-05:
```bash
$ git show
commit 6b65e63aa6ca2bb01cbca014984efad4ec131dda (HEAD -> master, origin/master, origin/HEAD)
Merge: 58947126f 7e60587a4
Author: Anton Kozlov <drakon.mega@gmail.com>
Date:   Tue Jul 4 12:35:38 2017 +0300

    Merge pull request #1055 from MaKToff/master
    
    Termios processing was separated from tty
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/7.1.1/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc-multilib/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++ --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --enable-multilib --disable-werror --enable-checking=release
Thread model: posix
gcc version 7.1.1 20170528 (GCC)
```
No extra CFLAGS in shell environment.

sikmir, 2017-07-05:
As strange as it may sound, it's caused by my locale (fi_FI.UTF-8). But it works with ru_RU.UTF-8. I don't really understand why.
```bash
$ echo $LANG
fi_FI.UTF-8
$ make distclean
$ make confload-x86/qemu
$ make
...
In file included from ./src/include/embox/test.h:32:0,
                 from src/tests/stdio/printf_test.c:10:
src/tests/stdio/printf_test.c: Funktio ”__test_case_at_line_113”:
src/tests/stdio/printf_test.c:122:23: virhe: ”snprintf” output truncated before the last format character [-Werror=format-truncation=]
  test_assert_equal(4, snprintf(dest, 4, "1234"));
./src/include/framework/test/assert.h:129:39: huom: in definition of macro ”test_assert_equal”
  __test_assertion_handle((actual) == (expected), \
                                       ^~~~~~~~
src/tests/stdio/printf_test.c:122:23: huom: ”snprintf” output 5 bytes into a destination of size 4
  test_assert_equal(4, snprintf(dest, 4, "1234"));
./src/include/framework/test/assert.h:129:39: huom: in definition of macro ”test_assert_equal”
  __test_assertion_handle((actual) == (expected), \
                                       ^~~~~~~~
src/tests/stdio/printf_test.c:125:23: virhe: ”12345678” directive output truncated writing 8 bytes into a region of size 2 [-Werror=format-truncation=]
  test_assert_equal(8, snprintf(dest, 2, "12345678"));
./src/include/framework/test/assert.h:129:39: huom: in definition of macro ”test_assert_equal”
  __test_assertion_handle((actual) == (expected), \
                                       ^~~~~~~~
src/tests/stdio/printf_test.c:125:23: huom: ”snprintf” output 9 bytes into a destination of size 2
  test_assert_equal(8, snprintf(dest, 2, "12345678"));
./src/include/framework/test/assert.h:129:39: huom: in definition of macro ”test_assert_equal”
  __test_assertion_handle((actual) == (expected), \
                                       ^~~~~~~~
cc1: all warnings being treated as errors
make[4]: *** [mk/image_lib.mk:33: build/base/obj/src/tests/stdio/printf_test.o] Virhe 1
make[3]: *** [mk/build.mk:23: build] Virhe 2
make[2]: *** [mk/load.mk:41: build] Virhe 2
make[1]: *** [mk/main.mk:30: build] Virhe 2
make: *** [Makefile:37: all] Virhe 2
$ LANG=en make
...
Build complete
$ LANG=ru make

...
Build complete
```

AntonKozlov, 2017-07-06:
Interesting, I was able to reproduce with `fi_FI.UTF-8` locale. With the locale, gcc invocation line is not contain `-Wno-error...` flags, but they are there with `ru_RU.UTF-8`, `en_US.UTF-8` locales that I could check. 
I suspect that gcc version determination is not working with your locale. I'll check shortly

PR #1095 (2017-07-06): (mk) Fix gcc version determination
Gcc should be called with --version flag to get machine-readable version
specifier.

Fixes #1093 

`gcc -v` gives hard-to-parse version format
```
$ GCC_VER="-v"; for loc in ru_RU fi_FI en_US; do echo == $loc; LANG=${loc}.UTF-8 gcc $GCC_VER 2>&1 | grep -e '^gcc' ; done
== ru_RU
gcc версия 7.1.1 20170528 (GCC) 
== fi_FI
gcc-versio 7.1.1 20170528 (GCC) 
== en_US
gcc version 7.1.1 20170528 (GCC) 
```

Using `gcc --version` gives better results
```
$ GCC_VER="--version"; for loc in ru_RU fi_FI en_US; do echo == $loc; LANG=${loc}.UTF-8 gcc $GCC_VER 2>&1 | grep -e  [... cut]
closed 2017-07-06 by sikmir