embox/embox #1106: platform/pjsip/stm32f4discovery not build in master

FateevAlexandr, 2017-08-02:
/tmp/cclophvh.s: Assembler messages:
/tmp/cclophvh.s:175: Error: selected processor does not support Thumb mode `vmrs r3,fpscr'
/tmp/cclophvh.s:184: Error: selected processor does not support Thumb mode `vmsr fpscr,r3'
/tmp/cclophvh.s:204: Error: selected processor does not support Thumb mode `vmrs r3,fpscr'
/tmp/cclophvh.s:213: Error: selected processor does not support Thumb mode `vmsr fpscr,r3'
Make.inc:144: recipe for target 'src/s_fma.c.o' failed
make[6]: *** [src/s_fma.c.o] Error 1
make[6]: Leaving directory '/home/gibabyte/STM32/embox/embox/build/extbld/third_party/lib/OpenLibmImpl/openlibm-2252cbd02c176a31f74fa10f8fcdf926c3124987'
Makefile:47: recipe for target '/home/gibabyte/STM32/embox/embox/build/extbld/third_party/lib/OpenLibmImpl/.builded' failed
make[5]: *** [/home/gibabyte/STM32/embox/embox/build/extbld/third_party/lib/OpenLibmImpl/.builded] Error 2
build/base/gen/third_party/lib/OpenLibmImpl.extbld_rule.mk:18: recipe for target 'build/base/gen/third_party/lib/OpenLibmImpl.__extbld-target' failed
make[4]: *** [build/base/gen/third_party/lib/OpenLibmImpl.__extbld-target] Error 2

What i do:
git clone https://github.com/embox/embox.git embox
cd embox
make confload-platform/pjsip/stm32f4discovery
make

$ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/home/gibabyte/STM32/gcc-arm-none-eabi-4_9-2014q4/bin/../lib/gcc/arm-none-eabi/4.9.3/lto-wrapper
Target: arm-none-eabi
Configured with: /home/build/work/GCC-4-9-build/src/gcc/configure --target=arm-none-eabi --prefix=/home/build/work/GCC-4-9-build/install-native --libexecdir=/home/build/work/GCC-4-9-build/install-native/lib --infodir=/home/build/work/GCC-4-9-build/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/home/build/work/GCC-4-9-build/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/home/build/work/GCC-4-9-build/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/home/build/work/GCC-4-9-build/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/home/build/work/GCC-4-9-build/install-native/arm-none-eabi --build=i686-linux-gnu --host=i686-linux-gnu --with-gmp=/home/build/work/GCC-4-9-build/build-native/host-libs/usr --with-mpfr=/home/build/work/GCC-4-9-build/build-native/host-libs/usr --with-mpc=/home/build/work/GCC-4-9-build/build-native/host-libs/usr --with-isl=/home/build/work/GCC-4-9-build/build-native/host-libs/usr --with-cloog=/home/build/work/GCC-4-9-build/build-native/host-libs/usr --with-libelf=/home/build/work/GCC-4-9-build/build-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Tools for ARM Embedded Processors' --with-multilib-list=armv6-m,armv7-m,armv7e-m,armv7-r
Thread model: single
gcc version 4.9.3 20141119 (release) [ARM/embedded-4_9-branch revision 218278] (GNU Tools for ARM Embedded Processors)

anton-bondarev, 2017-08-02:
@FateevAlexandr Thanks.

Please try arm-none-eabi version 5.4.1
```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1 20160919
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

FateevAlexandr, 2017-08-02:
Build with gcc version 5.4.1 20160919 (release)

closed 2017-08-02 by FateevAlexandr