embox/embox #1182: pjsip for stm32f4 doesn't compile

jonyMarino, 2018-07-20:
I compiled this way from master:
make confload-platform/pjsip/stm32f4discovery
make

Ang I got:

arm-none-eabi-gcc  -pipe --debug-prefix-map=`pwd`= --debug-prefix-map=./= -fno-strict-aliasing -fno-common -Wall -Werror -Wundef -Wno-trigraphs -Wno-char-subscripts -Wformat -std=gnu99 -Os -g -Wno-maybe-uninitialized -mthumb -mlittle-endian -march=armv7e-m -mcpu=cortex-m4 -msoft-float -ffreestanding -Wno-misleading-indentation -Wno-nonnull-compare -U__linux__ -Ulinux -U__linux -D__EMBOX__ -D__unix -imacros build/base/gen/config.lds.h -Ibuild/base/include -I./src/include -I./src/arch/arm/include -Ibuild/base/gen/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/arch/arm/include -I./platform/stm32f4/include -I./src/compat/linux/include -I./src/compat/posix/include -I./src/compat/libc/include -I/home/jonathan/workspace-mars-uml2ooc/embox/src/compat/cxx/include -nostdinc -MMD -MP  -include build/base/gen/include/config/embox/compat/posix/proc/vfork_entry.h -D__EMBUILD_MOD__=embox__compat__posix__proc__vfork_entry -c -o build/base/obj/src/compat/posix/proc/vfork.o src/compat/posix/proc/vfork.c
src/compat/posix/proc/vfork.c: In function 'vfork_body':
src/compat/posix/proc/vfork.c:44:40: error: 'struct task' has no member named 'parent'
  task_vfork = task_resource_vfork(child->parent);
                                        ^~

anton-bondarev, 2018-07-21:
Thanks for the report!
What is your compiler version?

anton-bondarev, 2018-07-21:
@jonyMarino Fixed in  #1183. But unfortunately I haven't a real board right now to check how it work. Check it please if you can.

closed 2018-10-15 by alexkalmuk