embox/embox #1205: Intentional or bug? : src/tests/stdio/printf_test.c:122:23: error: ‘snprintf’ output truncated before the last format character [-Werror=format-truncation=]

AnimusPEXUS, 2018-09-18:
```
gcc  -pipe --debug-prefix-map=`pwd`= --debug-prefix-map=./= -fno-strict-aliasing -fno-common -Wall -Werror -Wundef -Wno-trigraphs -Wno-char-subscripts -Wformat -std=gnu99 -O0 -gdwarf-2 -nostdinc -m32 -march=i386 -fno-stack-protector -Wno-array-bounds -Wno-misleading-indentation -Wno-nonnull-compare -U__linux__ -Ulinux -U__linux -D__EMBOX__ -D__unix -imacros build/base/gen/config.lds.h -Ibuild/base/include -I./src/include -I./src/arch/x86/include -Ibuild/base/gen/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/arch/x86/include  -I./src/compat/linux/include -I./src/compat/posix/include -I./src/compat/libc/include -I/home/agu/_local/p_c/github_embox_embox/src/compat/cxx/include -nostdinc -MMD -MP  -include build/base/gen/include/config/embox/test/stdio/printf_test.h -D__EMBUILD_MOD__=embox__test__stdio__printf_test -c -o build/base/obj/src/tests/stdio/printf_test.o src/tests/stdio/printf_test.c
In file included from ./src/include/embox/test.h:32,
                 from src/tests/stdio/printf_test.c:10:
src/tests/stdio/printf_test.c: In function ‘__test_case_at_line_113’:
src/tests/stdio/printf_test.c:122:23: error: ‘snprintf’ output truncated before the last format character [-Werror=format-truncation=]
  test_assert_equal(4, snprintf(dest, 4, "1234"));
./src/include/framework/test/assert.h:129:39: note: in definition of macro ‘test_assert_equal’
  __test_assertion_handle((actual) == (expected), \
                                       ^~~~~~~~
src/tests/stdio/printf_test.c:122:23: note: ‘snprintf’ output 5 bytes into a destination of size 4
  test_assert_equal(4, snprintf(dest, 4, "1234"));
./src/include/framework/test/assert.h:129:39: note: in definition of macro ‘test_assert_equal’
  __test_assertion_handle((actual) == (expected), \
                                       ^~~~~~~~
src/tests/stdio/printf_test.c:125:23: error: ‘12345678’ directive output truncated writing 8 bytes into a region of size 2 [-Werror=format-truncation=]
  test_assert_equal(8, snprintf(dest, 2, "12345678"));
./src/include/framework/test/assert.h:129:39: note: in definition of macro ‘test_assert_equal’
  __test_assertion_handle((actual) == (expected), \
                                       ^~~~~~~~
src/tests/stdio/printf_test.c:125:23: note: ‘snprintf’ output 9 bytes into a destination of size 2
  test_assert_equal(8, snprintf(dest, 2, "12345678"));
./src/include/framework/test/assert.h:129:39: note: in definition of macro ‘test_assert_equal’
  __test_assertion_handle((actual) == (expected), \
                                       ^~~~~~~~
cc1: all warnings being treated as errors
make[4]: *** [mk/image_lib.mk:33: build/base/obj/src/tests/stdio/printf_test.o] Error 1
make[3]: *** [mk/build.mk:23: build] Error 2
make[2]: *** [mk/load.mk:41: build] Error 2
make[1]: *** [mk/main.mk:30: build] Error 2
make: *** [Makefile:37: all] Error 2
```

anton-bondarev, 2018-09-18:
@AnimusPEXUS Thanks
What is your compiler version? and what template do you use?

AnimusPEXUS, 2018-09-18:
```
gcc --version
gcc (GCC) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
`make confload-x86/qemu`

anton-bondarev, 2018-09-18:
Now we don't support gcc 8. I hope we fix it soon.

AntonKozlov, 2018-09-18:
```
src/tests/stdio/printf_test.c:122:23: error: ‘snprintf’ output truncated before the last format character [-Werror=format-truncation=]
  test_assert_equal(4, snprintf(dest, 4, "1234"));
```
gcc issues warning as the result (in `dest`) do not hold all required bytes, deduced in static.
But the test is actually checks if snprintf works correctly in such situation. 
I think disabling the warning for this file with `#pragma`s would be good enough solution

closed 2018-10-16 by Deryugin

anton-bondarev, 2018-10-16:
@AnimusPEXUS @AntonKozlov Please check on your compiler

AnimusPEXUS, 2018-10-16:
both `make` and `make confload-x86/qemu` completed ok. the later printed out some cycle errors before main building process
```
>make
  CONFIGFILE conf/mods.config
  CONFIGLINK
  BUILDMODEL
  BUILDGEN build
 Error: modules dependency cycle:
         embox.compat.posix.proc.exit
         embox.compat.posix.proc.vfork_exchanged
 Error: modules dependency cycle:
         embox.mem.phymem
         embox.mem.vmem
 Error: modules dependency cycle:
         embox.mem.mmap_mmu
         embox.mem.vmem
 Error: modules dependency cycle:
         embox.mem.phymem
         embox.mem.vmem
 Error: modules dependency cycle:
         embox.mem.mmap_mmu
         embox.mem.vmem
 Error: modules dependency cycle:
         embox.compat.posix.proc.exit
         embox.compat.posix.proc.vfork_exchanged
 Error: modules dependency cycle:
         embox.mem.phymem
         embox.mem.vmem
 Error: modules dependency cycle:
         embox.mem.mmap_mmu
         embox.mem.vmem
 Error: modules dependency cycle:
         embox.compat.posix.proc.exit
         embox.compat.posix.proc.vfork_exchanged
 cp -r -T src/lib/debug/whereami.h build/base/include/debug/whereami.h
```

anton-bondarev, 2018-10-16:
@AnimusPEXUS Thanks
'dependency cycle' errors are internals (only for build debug) and don't affect runtime execution. We are cleaning them slowly.
Try execute on qemu please
`./scripts/qemu/auto_qemu`

AnimusPEXUS, 2018-10-16:
```
./scripts/qemu/auto_qemu
AUTOQEMU_ARCH: x86 (guessed)
AUTOQEMU_MEM: 256 (guessed)
AUTOQEMU_NICS: virtio (guessed)
AUTOQEMU_NICS_CONFIG: (got)
AUTOQEMU_AUDIO: (guessed)
AUTOQEMU_AUDIO_CONFIG: (got)
AUTOQEMU_GRAPHIC: (guessed)
AUTOQEMU_GRAPHIC_CONFIG: (got)
AUTOQEMU_DIAG: embox__driver__serial__i8250 (guessed)
AUTOQEMU_DIAG_CONFIG: (got)
no kvm kernel module loaded
AUTOQEMU_KVM_ARG: (guessed)
AUTOQEMU_LOAD_ARG: -kernel ./build/base/bin/embox (guessed)
AUTOQEMU_MACHINE: (guessed)

sudo qemu-system-i386 -kernel ./build/base/bin/embox -m 256 -net nic,vlan=0,model=virtio,macaddr=AA:BB:CC:DD:EE:02 -net tap,script=./scripts/qemu/start_script,downscript=./scripts/qemu/stop_script,vlan=0 -nographic
qemu-system-i386: Invalid parameter 'vlan'

```
```
qemu-system-i386 --version
QEMU emulator version 3.0.0
```

anton-bondarev, 2018-10-17:
@AnimusPEXUS Thanks a lot

Now I haven't qemu with version 3.0.0 and just open a new issue #1231  We try to do it as soon as possible

PR #1230 (2018-10-16): Add GCC8 compatibility
Fix build for x86 templates for GCC 8.2.0

Fix #1205