embox/embox #1335: execv return 0 and does not set errno on errors

anton-bondarev, 2019-04-16:
#1334 Improve `env` command to enable running program in modified environment show this bug

Ahish9009, 2020-02-24:
Hi,
Could I work on this issue?

Deryugin, 2020-02-25:
@Ahish9009 sure! Codestyle and git workflow can be found on wiki: https://github.com/embox/embox/wiki

Deryugin, 2020-02-29:
A little more details on this issue (as it seems to be a bit more complicated than it seemed before).

This due to `vfork()` implementation. `vfork()` is kind of optimization when you want to do `fork()` just to `execv()` in the child process without actual duplication of all process resources. The first problem raises because we "disconnect" child from the parent too early, before checking that the command really exists, so instead of returning error code for `execv()` child process just exits with this code. The second problem is that `execv()` is not checking for return values of function that it uses, so it simply returns zero.

Ahish9009, 2020-02-29:
Hi

Thanks for the additional information!

Deryugin, 2020-03-10:
@Ahish9009 how is it going?

Ahish9009, 2020-03-10:
Hi
I’ve been understanding the code base for most of the past days, have began working on this yesterday, will update soon.

Ahish9009, 2020-03-13:
Hey
So if I am understanding what's happening correctly, the vfork implementation currently copies and saves register values, and then calls the vfork_start_child function. I'm not sure where it is calling this function from as it is defined in 3 places:
- vfork_exchanged
- vfork_none
- vfork_stop_parent
A little more clarity on this would be really helpful!

anton-bondarev, 2020-03-14:
@Ahish9009 I think vfork_exchanged implementation is used in your template. You can check it if make a compilation mistake in src/compat/posix/proc/vfork_exchange.c
I think there are following lines in your ./conf/mods.config
```
	include embox.compat.posix.proc.vfork_exchanged
	include embox.compat.posix.proc.exec_exchanged
```

PR #1867 (2020-03-19): execv: fix execv returns 0 and doesn't set errno
Closes #1335:
Behaves like execv in posix: returns -1 on failure and sets errno to the value of error that occurred. Seems to have fixed #1334, but does not seem to have helped #1702.
closed 2020-03-19 by anton-bondarev