embox/embox #1458: Can't build for platform/stellaris/lm3s811evb

Dovgalyuk, 2019-07-26:
Got the following errors:

1
src/drivers/serial/pl011/pl011.c:101:40: error: 'pl011_mem' defined but not used [-Werror=unused-const-variable=]
 static const struct periph_memory_desc pl011_mem = {
cc1: all warnings being treated as errors

This one is solved by patching memory.h: 
#define PERIPH_MEMORY_DEFINE(_mem_desc) const void * const p##_mem_desc = &_mem_desc;

2
./src/include/kernel/thread.h:18:33: fatal error: kernel/thread/types.h: No such file or directory
 #include <kernel/thread/types.h>

Have no idea about this one.

arm-none-eabi-gcc is installed from Ubuntu-18.04 repository.
gcc version 6.3.1 20170620 (15:6.3.1+svn253039-1build1)

anton-bondarev, 2019-07-26:
Thanks!
Try again in master, please. We've fixed stellaris/lm3s811evb build in PR #1459 
[image]
But the first problem

>  src/drivers/serial/pl011/pl011.c:101:40: error: 'pl011_mem' defined but not used [-Werror=unused-const-variable=])

 hasn't appeared.

Dovgalyuk, 2019-07-26:
It's latest master already. Nothing has changed.
I used make confload-platform/stellaris/lm3s811evb

This warning looks correct, because the variable is not actually used. Therefore the code should be fixed.

anton-bondarev, 2019-07-26:
Are you sure that your git branch has changes from PR #1459 ?

After pulled git you have to load the config again
`make confload-platform/stellaris/lm3s811evb`
and may be try cleaning project
`make clean cacheclean`
than build
`make`

What is you arm compiler version?

Dovgalyuk, 2019-07-26:
Yes, I've updated the branch.

Please check the first message for the compiler version.

anton-bondarev, 2019-07-26:
> Please check the first message for the compiler version.
Ops, sorry, I haven't noticed.

Could you give a full build log?
Do `make distclean` before building.
After building pack your ./conf folder to a tarball and send it

Dovgalyuk, 2019-07-26:
MKGEN mk_core_def.mk
 MKGEN mk_core_obj.mk
 MKGEN mk_util.mk
 MKGEN mk_gold_engine.mk
 MKGEN mk_model.mk
 MKGEN mk_annotations_core.mk
 MKGEN mk_mybuild_myfile.mk
 MKGEN mk_mybuild_configfile.mk
 MKGEN mk_mybuild_build.mk
 MKGEN mk_mybuild.mk
 MKGEN mk_annotations_handlers.mk
 MKGEN mk_ugly.mk
 ANNOTATION CORE
 MYFILE mk/mybuild/lang/Unique.my
 MYFILE mk/mybuild/lang/DefineMacro.my
 MYFILE mk/mybuild/lang/Generated.my
 MYFILE mk/mybuild/lang/Runlevel.my
 MYFILE mk/mybuild/lang/Cflags.my
 MYFILE mk/mybuild/lang/Build.my
 MYFILE mk/mybuild/lang/WithTest.my
 MYFILE mk/mybuild/lang/WithAllTests.my
 MYFILE mk/mybuild/lang/DefaultImpl.my
 MYFILE mk/mybuild/lang/Cmd.my
 MYFILE mk/mybuild/lang/Postbuild.my
 MYFILE mk/mybuild/lang/Rule.my
 MYFILE mk/mybuild/lang/InitFS.my
 MYFILE mk/mybuild/lang/NoRuntime.my
 MYFILE mk/mybuild/lang/Mandatory.my
 MYFILE mk/mybuild/lang/Include.my
 MYFILE mk/mybuild/lang/For.my
 MYFILE mk/mybuild/lang/InstrumentProfiling.my
 MYFILE mk/mybuild/lang/IncludePath.my
 MYFILE mk/mybuild/lang/AddPrefix.my
 MYFILE mk/mybuild/lang/WithValue.my
 MYFILE mk/mybuild/lang/App.my
 MYFILE mk/mybuild/lang/AutoCmd.my
 MYFILE mk/mybuild/lang/NumConstraint.my
 MYFILE mk/mybuild/lang/Type.my
 MYFILE mk/mybuild/lang/TestFor.my
 MYFILE mk/mybuild/lang/LinkerSection.my
 MYFILE mk/mybuild/lang/IncludeExport.my
 MYFILE mk/mybuild/lang/IfNeed.my
 MYFILE mk/mybuild/lang/Module.my
 MYFILE mk/mybuild/lang/BuildDepends.my
 MYFILE mk/mybuild/lang/BuildArtifactPath.my
 MYFILE src/lib/edit/Mybuild
 MYFILE src/lib/exec/Mybuild
 MYFILE src/lib/elf/Mybuild
 MYFILE src/lib/crypt/Mybuild
 MYFILE src/lib/execinfo/Mybuild
 MYFILE src/lib/readline/Mybuild
 MYFILE src/lib/libgcc/Mybuild
 MYFILE src/lib/tftp/Mybuild
 MYFILE src/lib/Mybuild
 MYFILE src/lib/fps/Mybuild
 MYFILE src/lib/debug/Mybuild
 MYFILE src/lib/shell/Mybuild
 MYFILE src/lib/stl/Mybuild
 MYFILE src/lib/bmp/Mybuild
 MYFILE src/lib/libsupcxx/Mybuild
 MYFILE src/util/bitmap/Mybuild
 MYFILE src/util/hashtable/Mybuild
 MYFILE src/util/Mybuild
 MYFILE src/util/dlist/Mybuild
 MYFILE src/util/bit/Mybuild
 MYFILE src/util/array_spread/Mybuild
 MYFILE src/kernel/cpu/Mybuild
 MYFILE src/kernel/cpu/Smp.my
 MYFILE src/kernel/lthread/sync/Mybuild
 MYFILE src/kernel/lthread/Mybuild
[... 735 lines cut ...]
arm-none-eabi-ld -r -o build/base/obj/module/embox/arch/generic/arch.o -N -g \
	./build/base/obj/src/arch/generic/arch.o
arm-none-eabi-gcc  -pipe --debug-prefix-map=`pwd`= --debug-prefix-map=./= -fno-strict-aliasing -fno-common -Wall -Werror -Wundef -Wno-trigraphs -Wno-char-subscripts -Wformat -std=gnu99 -Os -g -mthumb -mlittle-endian -march=armv7-m -mcpu=cortex-m3 -ffreestanding -msoft-float -Wno-misleading-indentation -Wno-nonnull-compare -U__linux__ -Ulinux -U__linux -D__EMBOX__ -D__unix -imacros build/base/gen/config.lds.h -Ibuild/base/include -I./src/include -I./src/arch/arm/include -Ibuild/base/gen/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/arch/arm/include -I./platform/lm3s811evb/include -I./src/compat/linux/include -I./src/compat/posix/include -I./src/compat/libc/include -I/hdd/work/embox/src/compat/cxx/include -nostdinc -MMD -MP  -include build/base/gen/include/config/embox/driver/interrupt/cortexm_nvic.h -D__EMBUILD_MOD__=embox__driver__interrupt__cortexm_nvic -c -o build/base/obj/src/drivers/interrupt/cortexm/cortexm_nvic.o src/drivers/interrupt/cortexm/cortexm_nvic.c
arm-none-eabi-gcc  -pipe --debug-prefix-map=`pwd`= --debug-prefix-map=./= -Os -g -mthumb -mlittle-endian -march=armv7-m -mcpu=cortex-m3 -ffreestanding -msoft-float -U__linux__ -Ulinux -U__linux -D__EMBOX__ -D__unix -imacros build/base/gen/config.lds.h -Ibuild/base/include -I./src/include -I./src/arch/arm/include -Ibuild/base/gen/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/arch/arm/include -I./platform/lm3s811evb/include -I./src/compat/linux/include -I./src/compat/posix/include -I./src/compat/libc/include -I/hdd/work/embox/src/compat/cxx/include -nostdinc -MMD -MP  -include build/base/gen/include/config/embox/driver/interrupt/cortexm_nvic.h -D__EMBUILD_MOD__=embox__driver__interrupt__cortexm_nvic -c -o build/base/obj/src/drivers/interrupt/cortexm/cortexm_irq_handle.o src/drivers/interrupt/cortexm/cortexm_irq_handle.S
arm-none-eabi-ld -r -o build/base/obj/module/embox/driver/interrupt/cortexm_nvic.o -N -g \
	./build/base/obj/src/drivers/interrupt/cortexm/cortexm_nvic.o \
	./build/base/obj/src/drivers/interrupt/cortexm/cortexm_irq_handle.o
arm-none-eabi-gcc  -pipe --debug-prefix-map=`pwd`= --debug-prefix-map=./= -fno-strict-aliasing -fno-common -Wall -Werror -Wundef -Wno-trigraphs -Wno-char-subscripts -Wformat -std=gnu99 -Os -g -mthumb -mlittle-endian -march=armv7-m -mcpu=cortex-m3 -ffreestanding -msoft-float -Wno-misleading-indentation -Wno-nonnull-compare -U__linux__ -Ulinux -U__linux -D__EMBOX__ -D__unix -imacros build/base/gen/config.lds.h -Ibuild/base/include -I./src/include -I./src/arch/arm/include -Ibuild/base/gen/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/arch/arm/include -I./platform/lm3s811evb/include -I./src/compat/linux/include -I./src/compat/posix/include -I./src/compat/libc/include -I/hdd/work/embox/src/compat/cxx/include -nostdinc -MMD -MP  -include build/base/gen/include/config/embox/driver/clock/cortexm_systick.h -D__EMBUILD_MOD__=embox__driver__clock__cortexm_systick -c -o build/base/obj/src/drivers/clock/cortexm_systick.o src/drivers/clock/cortex [... cut ...]
cc1: all warnings being treated as errors
mk/image_lib.mk:33: recipe for target 'build/base/obj/src/drivers/serial/pl011/pl011.o' failed
make[4]: *** [build/base/obj/src/drivers/serial/pl011/pl011.o] Error 1
mk/build.mk:19: recipe for target 'build' failed
make[3]: *** [build] Error 2
mk/load.mk:41: recipe for target 'build' failed
make[2]: *** [build] Error 2
mk/main.mk:30: recipe for target 'build' failed
make[1]: *** [build] Error 2
Makefile:37: recipe for target 'all' failed
make: *** [all] Error 2

Dovgalyuk, 2019-07-26:
[conf.tar.gz](https://github.com/embox/embox/files/3435693/conf.tar.gz)

anton-bondarev, 2019-07-26:
@Dovgalyuk Thank you very much!
I'm sorry, I didn't understand you at first.
You are right. The problem was in PERIPH_MEMORY_DEFINE macro. I hope I've fixed it in #1460.

Could you check it again?

Dovgalyuk, 2019-07-29:
Now the build is ok.

closed 2019-07-29 by Dovgalyuk