embox/embox #1537: usermode86/debug doesn't restore terminal settings

AntonKozlov, 2019-10-30:
After (forceful) exit from usermode86/debug, terminal settings are not restored
```
$ make confload-usermode86/debug
$ make
$ ./build/base/bin/embox
...
CTRL+C
```

Ahish9009, 2020-02-23:
Hi, I would like to work on this issue.
I have built the embox shell, could you point to me a good place to get started to resolve this issue?

bartekxk, 2020-02-24:
Hello,
What settings are not properly restored? I'm trying to reproduce the error but everything looks fine.

Thanks.

AntonKozlov, 2020-02-24:
I filed the bug after conversation with @anton-bondarev. 

AFAIR, after embox forcefully terminated (i.e. ctrl-c or with `kill` from another terminal on linux), terminal doesn't do echo of typed symbols, enter key doesn't introduce new line (or makes '\n' without '\r', don't remember now)

worldinmyfist, 2020-02-24:
Hi, Can I work on this issue ?

Ahish9009, 2020-02-24:
Hi
I am also unable to reproduce the error. I tried echo'ing various symbols, and using the enter key. Everything seems to be working fine for me.

AntonKozlov, 2020-02-24:
Please check screencast with the issue demonstrated.
Is behavior on your machine different?

https://asciinema.org/a/304411

I hit ctrl-c during boot process. Then I hit enter few times -- shell invitations expected on separate lines, but they are on the same. 
Then I tried typing `1<enter>2<enter>...`. I expected to see number as I type them, but they are not displayed -- I see only shell error after the input processed. 
Typing `reset` blindly resets the terminal. 
Just for reference, correct behavior for same keys pressed https://asciinema.org/a/304416

Ahish9009, 2020-02-24:
[image]

It seems to be working fine on my machine.

AntonKozlov, 2020-02-25:
I see, interesting. I use xterm on linux, it probably matters. It may be my
environment as well.
I would suggest not to spend too much efforts on reproducing, it's warmup
issue and it's not very disturbing. Thanks for checking! I think it made a
contribution to understanding of the problem the problem.

anton-bondarev, 2020-02-25:
@Ahish9009 @bartekxk @AntonKozlov Thanks
Now I also can't reproduce the problem!

closed 2020-02-25 by anton-bondarev