embox/embox #1569: Tish: Arrow buttons error

HDDCableHD, 2019-11-26:
Left and Right arrow buttons don't work.
Up and Down keys work correctly.

Deryugin, 2019-11-29:
Please, provide more details. Which configuration do you use?

AntonKozlov, 2019-11-29:
I've asked @HDDCableHD (Georgy Doronin, not sure about how it's written) to make this entry. He reproduced the issue. So it should be `x86/qemu`

AntonKozlov, 2019-12-17:
@HDDCableHD made some analysis and suggest using a next patch https://github.com/HDDCableHD/embox/commit/4bd6fd99cc577aaf085fb08fd71f1772dfc65846#diff-27f710a4260ba96f651c03cc0b17055f

it also appears like 55c225ad39 introduced the bug.

But there is an another issue: backspacing in the middle of the line do not moves the rest of the line.

Reverting 55c225ad39 fixes original issue but do not fixes backspacing in the middle. The suggested patch fixes both. 

@embox/devs please a feedback about the patch. 55c22 avoided control sequences for some reason. The patch introduces ones. It's possible than reverting 55c22 is another way, with fixing backspacing another way

HDDCableHD, 2019-12-24:
If unlock these lines, the backspace problem will be solved: [L583-L585](https://github.com/HDDCableHD/embox/blob/master/third-party/lib/linenoise/linenoise.c#L583-L585).

AntonKozlov, 2020-01-03:
> If unlock these lines, the backspace problem will be solved: [L583-L585](https://github.com/HDDCableHD/embox/blob/master/third-party/lib/linenoise/linenoise.c#L583-L585).

Thanks! It seems like both issues can be fixed by reverting few changes.

@Deryugin, commit 4928d40d introduced the issue. Can we revert it?

@HDDCableHD could you prepare a pull request with fixes? I would prefer to have revert commits (`git revert ...`) in the PR

anton-bondarev, 2020-02-28:
@HDDCableHD @AntonKozlov Thanks.
I think that the right way to solve the problem is pulling the last version of [linenoise](https://github.com/antirez/linenoise).

@HDDCableHD Are you still working on the issue?

ThePiGrepper, 2020-03-05:
> @HDDCableHD @AntonKozlov Thanks.
> I think that the right way to solve the problem is pulling the last version of [linenoise](https://github.com/antirez/linenoise).
> 
> @HDDCableHD Are you still working on the issue?

hi. is there still interest in upgrading linenoise to its latest version? I could do it if you like.

anton-bondarev, 2020-03-05:
> hi. is there still interest in upgrading linenoise to its latest version? I could do it if you like.

Yes, we still interest in the latest version.

Be careful. it 's not usual a third-party application. It's placed directlyin Embox sources [https://github.com/embox/embox/tree/master/src/lib/readline](https://github.com/embox/embox/tree/master/src/lib/readline)

AntonKozlov, 2020-03-05:
> hi. is there still interest in upgrading linenoise to its latest version? I could do it if you like.

I don't think upgrade is correct way to solve the issue. 

The issue narrowed down till a single (our) commit changing linenoise behavior. It's possible to just revert one commit and consider this is resolved.

I don't understand at all why don't take this path. The suggestion to upgrade will revert all changes without considering each and bring new linenoise for no reason. Have anyone tested it?

I'm not opposite linenoise upgrading, but it would be a little relevant for this issue.

This isuue at this moment is more about communication and policy: when we patch third-party and how we deal with regressions

anton-bondarev, 2020-03-05:
@ThePiGrepper @AntonKozlov  We need time for discuss and investigate this issue

PR #1946 (2020-04-20): third-party: Porting new version of linenoise
Closes #1946
It also fixes #1569
closed 2020-04-20 by anton-bondarev