embox/embox #1644: bsd/queue support

Deryugin, 2020-01-21:
Right now we don't have any modules for bsd compatibility, I suggest creating a separate directory for it, like `src/compat/bsd/`.

We need to implement macros/functions from BSD `queue.h`, it can be tested with #1643

AntonKozlov, 2020-01-21:
Can we just take them as is
https://github.com/freebsd/freebsd/blob/master/sys/sys/queue.h ? It's BSD
too

src/compat/bsd is nice place, but on my Linux system it's just in
/usr/include
```
$ head /usr/include/sys/queue.h
/*
 * Copyright (c) 1991, 1993
 *      The Regents of the University of California.  All rights reserved.
 *
```

Deryugin, 2020-01-21:
Yeah, probably we can just take this file.

As for the header directory, we're trying to avoid putting header files into `src/include` by using `@IncludeExport`, otherwise `src/include` directory becomes a bit messy.

subhojit-ghorai, 2020-03-02:
Hi! Can I work on this issue? I have been doing research on this issue.

Deryugin, 2020-03-02:
Hi, 
Sure!

commit 2fd5f0ff (2020-03-16): Merge pull request #1784 from subhojit-ghorai/FIXED-#1644

implemented macros/functions from BSD queue.h. (Fixed #1644)
closed 2020-03-16 by Deryugin