embox/embox #1645: Extend pthread compatibility

Deryugin, 2020-01-21:
We need to implement following functions:
```
int pthread_barrier_destroy(pthread_barrier_t );
int pthread_barrier_init(pthread_barrier_t restrict, const pthread_barrierattr_t restrict, unsigned);
int pthread_barrier_wait(pthread_barrier_t );
int pthread_barrierattr_destroy(pthread_barrierattr_t *);
```
This changes are required for #1643

sobhaniiest, 2020-03-08:
Hi, I am interested to work on the issue [npf porting #1643](https://github.com/embox/embox/issues/1643) as this one is the required changes for #1643, I have done some study about how pthread_barrier functions are implemented in pthreads-win32, QNX and Glibc. Could anyone please assign me this issue? And is there any module that fetch and add or subtract a variable atomically in embox? 
Thanks :)

Deryugin, 2020-03-09:
Hi @sobhaniiest 
Those implementations (pthreads-win32, QNX and Glibc) are implementations of POSIX standard; refer to https://pubs.opengroup.org/onlinepubs/9699919799/ for details on how exactly should those functions behave.

sobhaniiest, 2020-03-12:
Hi, I have implemented pthread_barrier functions along with its test program. Please review these.
Thanks :)