embox/embox #1647: devfs/ls fails on arm/qemu

Deryugin, 2020-01-21:
Bug can be reproduced on arm/qemu template.

You should add lots of lines with `"ls /dev/ttyS0",` to `system_start.inc`. After a few dozens of successfull calls, `ls` stalls.

layman-n-ish, 2020-03-03:
Hello! I tried reproducing this bug on `arm/qemu` template by appending `"ls /dev/ttyS0",` (~500 times!) to `system_start.inc`, as suggested, although I don't see any signs of `ls` stalling. Was the bug fixed already?

anton-bondarev, 2020-03-04:
What your qemu version?

layman-n-ish, 2020-03-04:
```
$ qemu-system-arm --version

QEMU emulator version 4.0.0 (Debian 1:4.0+dfsg-0ubuntu9.4)
Copyright (c) 2003-2019 Fabrice Bellard and the QEMU Project developers
```

anton-bondarev, 2020-03-04:
I also have not been able to reproduced the problem.
@Deryugin How many lines with `"ls /dev/ttyS0",` should be added?

Deryugin, 2020-03-04:
~50 calls were enough to fail.

Now I can't reproduce it too, so I close this issue.

```
> qemu-system-arm --version
QEMU emulator version 4.2.0
Copyright (c) 2003-2019 Fabrice Bellard and the QEMU Project developers
> arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/9.2.0/lto-wrapper
Target: arm-none-eabi
Configured with: /build/arm-none-eabi-gcc/src/gcc-9.2.0/configure --target=arm-none-eabi --prefix=/usr --with-sysroot=e
Thread model: single
gcc version 9.2.0 (Arch Repository) 
```

closed 2020-03-04 by Deryugin