embox/embox #1653: No output with sifive_u qemu 4.2.0

alexkalmuk, 2020-01-28:
template: `riscv/qemu-sifive-u`
qemu:
```
$ qemu-system-riscv32 --version
QEMU emulator version 4.2.0
Copyright (c) 2003-2019 Fabrice Bellard and the QEMU Project developers
```

Run embox:
```
$ qemu-system-riscv32 -nographic -machine sifive_u -kernel embox
qemu-system-riscv32: warning: No -bios option specified. Not loading a firmware.
qemu-system-riscv32: warning: This default will change in a future QEMU release. Please use the -bios option to avoid breakages when this happens.
qemu-system-riscv32: warning: See QEMU's deprecation documentation for details.
```

And.. no more output here.

P.S. But works well with qemu 4.1.0.

alexkalmuk, 2020-01-29:
But template `riscv/qemu` works on qemu 4.2.0.

anton-bondarev, 2020-02-29:
@nastya-nizharadze As I understand qemu-system-riscv32 with -machine sifive_u version 4.1.0 has wrong address map?

nastya-nizharadze, 2020-03-06:
In qemu version 4.1.0 for the sifive_u machine, the uart address was not the same as that specified in the documentation of the machine emulated by it. In qemu version 4.2.0, this issue has been fixed. After fixing this address in embox, another problem became clear: in qemu 4.2.0 for the sifive_u machine, smp cpus = 2 and it is impossible to do less, but I did not handle this situation, due to which the output was incorrect.

PR #1787 (2020-03-06): riscv: add multiple harts case handling
fix #1653
closed 2020-03-06 by alexkalmuk