embox/embox #1654: risc-v version command wrong output

alexkalmuk, 2020-01-28:
template: riscv/qemu-sifive-u
qemu: 4.1.0

```
$qemu-system-riscv32 -nographic -machine sifive_u -kernel embox
***
root@embox:/#version                                                            

.------.          _
|  ____|         | | 
| |____ _ __ ___ | |_    _____  __
|  ____| '_ ` _ \|  _ \ / _ \ \/ /
| |____| | | | | | |_) | (_) >  <  
|______|_| |_| |_|____/ \___/_/\_\

Tue Jan 28 18:30:00 MSK 2020

Compiler: 8.2.0
```

On host:
```
$ date
Tue Jan 28 19:24:41 MSK 2020
```

But `version` works fine on x86 and arm.

alexkalmuk, 2020-01-28:
`18:30:00` constantly appears and doesn't depend on the actual build time.

zhuanyanchengkong, 2020-03-01:
I want to try to solve this．

alexkalmuk, 2020-03-01:
@zhuanyanchengkong Let's start with working on one single issue. Should I unassign you from https://github.com/embox/embox/issues/1720?

zhuanyanchengkong, 2020-03-02:
> @zhuanyanchengkong Let's start with working on one single issue. Should I unassign you from #1720?

Sorry to bother you, I don't know if I haven't finished issue #1720. I think I should complete issue # 1720 first. You can unassign me from #1654.

anton-bondarev, 2020-03-02:
No, no problem for #1720 . It's really difficult to starting, we wanted to try different type (research)  issue :)

Try to ask @nastya-nizharadze . She ported Embox to RISC-V

zhuanyanchengkong, 2020-03-03:
> No, no problem for #1720 . It's really difficult to starting, we wanted to try different type (research) issue :)
> 
> Try to ask @nastya-nizharadze . She ported Embox to RISC-V

OK, thank you, I understand. I also spend some time using GBD, tracking error messages, and trying to locate error codes.

anton-bondarev, 2020-03-03:
@zhuanyanchengkong Ok.
 But In my mind the problem can be solve in different way:)

zhuanyanchengkong, 2020-03-03:
I am so sorry. I may have to abandon this issue because we delayed school due to pneumonia and there is no risc-v board at home.

anton-bondarev, 2020-03-03:
@zhuanyanchengkong 
Try to get  `info mtree` in qemu control console :)

zhuanyanchengkong, 2020-03-03:
> @zhuanyanchengkong
> Try to get `info mtree` in qemu control console :)

Ok!

anton-bondarev, 2020-03-05:
> Try to get `info mtree` in qemu control console

Did you reproduced the problem?
You should run Embox on  qemu-system-riscv32 different version (4.1 and 4.2) and compare `info mem` in them

zhuanyanchengkong, 2020-03-05:
> Did you reproduced the problem?
> You should run Embox on qemu-system-riscv32 different version (4.1 and 4.2) and compare `info mem` in them

Sry. I am still solving this compilation problem. 

>  CONFIGFILE conf/mods.config
 CONFIGLINK
 BUILDMODEL
 BUILDGEN build
cp -r -T src/lib/debug/whereami.h build/base/include/debug/whereami.h
cp -r -T src/mem/vmem/vmem_nommu.h build/base/include/mem/vmem.h
cp -r -T src/drivers/diag/diag.h build/base/include/drivers/diag.h
cp -r -T src/compat/libc/math/math_builtins.h build/base/include//math.h
cp -r -T src/util/log.h build/base/include/util/log.h
cp -r -T src/util/logging.h build/base/include/util/logging.h
cp -r -T src/profiler/tracing/trace.h build/base/include/profiler/tracing/trace.h
cp -r -T src/drivers/tty/termios_ops.h build/base/include/drivers/tty/termios_ops.h
cp -r -T src/kernel/lthread/lthread.h build/base/include/kernel/lthread/lthread.h
cp -r -T src/kernel/lthread/lthread_sched_wait.h build/base/include/kernel/lthread/lthread_sched_wait.h
cp -r -T src/kernel/lthread/waitq.h build/base/include/kernel/lthread/waitq.h
cp -r -T src/kernel/thread/thread_types.h build/base/include/kernel/thread/types.h
make[4]: Nothing to be done for 'all'.
make[4]: '/home/eric/Project/embox-0.4.0/build/base/gen/embox_gcc_env.sh' is up to date.
riscv64-unknown-elf-gcc -E  -P -undef -D__LDS__ -U__linux__ -Ulinux -U__linux -D__EMBOX__ -D__unix -imacros build/base/gen/config.lds.h -Ibuild/base/include -I./src/include -I./src/arch/riscv/include -Ibuild/base/gen/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/arch/riscv/include  -I./src/compat/linux/include -I./src/compat/posix/include -I./src/compat/libc/include -I/home/eric/Project/embox-0.4.0/src/compat/cxx/include -nostdinc -MMD -MP  -include build/base/gen/include/config/embox/kernel/stack.h -D__EMBUILD_MOD__=embox__kernel__stack \
-Ibuild/base/gen \
-imacros build/base/gen/config.lds.h \
	-MMD -MT build/base/obj/src/kernel/stack.lds -MF build/base/obj/src/kernel/stack.lds.d -o build/base/obj/src/kernel/stack.lds src/kernel/stack.lds.S
make[4]: riscv64-unknown-elf-gcc: Command not found
mk/image_lib.mk:46: recipe for target 'build/base/obj/src/kernel/stack.lds' failed
make[4]: *** [build/base/obj/src/kernel/stack.lds] Error 127
mk/build.mk:19: recipe for target 'build' failed
make[3]: *** [build] Error 2
mk/load.mk:41: recipe for target 'build' failed
make[2]: *** [build] Error 2
mk/main.mk:30: recipe for target 'build' failed
make[1]: *** [build] Error 2
Makefile:37: recipe for target 'all' failed
make: *** [all] Error 2

zhuanyanchengkong, 2020-03-05:
`make[4]: riscv64-unknown-elf-gcc: Command not found` 
In my terminal I can use this command: ` riscv64-unknown-elf-gcc` and I can found it in $PATH. But It still not found when I build embox.

anton-bondarev, 2020-03-05:
@zhuanyanchengkong 
what output from
```riscv64-unknown-elf-gcc --version```

zhuanyanchengkong, 2020-03-05:
> riscv64-unknown-elf-gcc (GCC) 9.2.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions. There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR `PURPOSE`

zhuanyanchengkong, 2020-03-05:
Maybe the compilation chain is not installed properly. I am recompiling riscv-unknow-elf-gcc.

anton-bondarev, 2020-03-05:
have you  riscv-unknow-elf-gcc not riscv64-unknown-elf-gcc?
You can just change CROSS_COMPILER in conf/build.conf

zhuanyanchengkong, 2020-03-06:
Ok, I understand． @anton-bondarev

zhuanyanchengkong, 2020-03-06:
@alexkalmuk Hi,  @alexkalmuk  When I use the command `sudo ./scripts/qemu/auto_qemu `, I cannot enter embox.
> eric@Thinkpad-T480s ~/P/E/r/embox-0.4.0> sudo ./scripts/qemu/auto_qemu                                                                                                        
AUTOQEMU_ARCH: riscv (guessed)
AUTOQEMU_MEM: 512 (guessed)
AUTOQEMU_NICS: (guessed)
AUTOQEMU_NICS_CONFIG: (got)
AUTOQEMU_AUDIO: (guessed)
AUTOQEMU_AUDIO_CONFIG: (got)
AUTOQEMU_USB: (guessed)
AUTOQEMU_USB_CONFIG: (got)
AUTOQEMU_GRAPHIC: (guessed)
AUTOQEMU_GRAPHIC_CONFIG: (got)
AUTOQEMU_DIAG: embox__driver__serial__sifive_uart (guessed)
AUTOQEMU_DIAG_CONFIG: (got)
AUTOQEMU_KVM_ARG: (guessed)
AUTOQEMU_LOAD_ARG: -kernel ./build/base/bin/embox (guessed)
AUTOQEMU_MACHINE: -M virt (guessed)
qemu-system-riscv32 -M virt -kernel ./build/base/bin/embox -m 512 -nographic

 I found that many services were missing from the mods.config and no system startup items were added. I tried to add the telnet service, but it failed. How do you do it?

alexkalmuk, 2020-03-06:
Hi @zhuanyanchengkong, please follow the instruction from the issue description, there is no `auto_qemu` there

zhuanyanchengkong, 2020-03-06:
> Hi @zhuanyanchengkong, please follow the instruction from the issue description, there is no `auto_qemu` there

Yes,  I have tried both ways．

> eric@Thinkpad-T480s ~/P/E/r/e/b/b/bin> sudo qemu-system-riscv32 -nographic -machine sifive_u -kernel embox                                                                    (base) 

Embox kernel start
runlevel: init level is 0
runlevel: init level is 1
	unit: initializing embox.kernel.task.task_resource: done
	unit: initializing embox.driver.interrupt.riscv_intc: done
	unit: initializing embox.kernel.task.kernel_task: done
	unit: initializing embox.arch.riscv.kernel.interrupt: done
	unit: initializing embox.kernel.sched.sched: done
	unit: initializing embox.driver.clock.riscv_clk: done
	unit: initializing embox.kernel.time.kernel_time: done
	unit: initializing embox.kernel.time.timer_handler: done
	test: running embox.test.stdlib.setjmp_test . done
	unit: initializing embox.kernel.sched.sched_ticker_preempt: done
runlevel: init level is 2
runlevel: init level is 3
runlevel: init level is 4

alexkalmuk, 2020-03-06:
I run it as following:

```
qemu-system-riscv32 -nographic -machine sifive_u -kernel build/base/bin/embox
```

Can you please tell `qemu-system-riscv32 --version`

zhuanyanchengkong, 2020-03-06:
> eric@Thinkpad-T480s ~/P/E/r/e/b/b/bin> sudo qemu-system-riscv32 --version                                                                                                     (base) 
[sudo] password for eric: 
QEMU emulator version 4.0.50 (v4.0.0-1854-g57dfc2c4d5-dirty)
Copyright (c) 2003-2019 Fabrice Bellard and the QEMU Project developers

alexkalmuk, 2020-03-06:
And template is `riscv/qemu-sifive-u`? You didn't not modify it?

```
Embox kernel start
runlevel: init level is 0
runlevel: init level is 1
unit: initializing embox.kernel.task.task_resource: done
unit: initializing embox.driver.interrupt.riscv_intc: done
unit: initializing embox.kernel.task.kernel_task: done
unit: initializing embox.arch.riscv.kernel.interrupt: done
unit: initializing embox.kernel.sched.sched: done
```

It looks like embox is running. Do you type something in Embox's shell?

zhuanyanchengkong, 2020-03-06:
> It looks like embox is running. Can you type something in Embox's shell?

No, I can't do anything.

alexkalmuk, 2020-03-06:
Well, it could be happened probably due to some qemu bugs because RISC-V is a relatively recently added to qemu. So maybe versions under 4.2.0 can accidentally suffer from drawbacks... This issue also is not reproducible anymore.

@zhuanyanchengkong Are you interested to continue with RISC-V? If so we can propose you another simple task, but already under qemu 4.2.0 (so you have to install qemu 4.2.0)

zhuanyanchengkong, 2020-03-06:
@alexkalmuk OK, I'll try again, if it doesn't work, I will consider other issues．

alexkalmuk, 2020-03-06:
@zhuanyanchengkong Great! Please take a look at https://github.com/embox/embox/issues/1791, it is almost the same (same template), but you have to install qemu 4.2.0 first.

zhuanyanchengkong, 2020-03-06:
Ok!

zhuanyanchengkong, 2020-03-10:
Hi, I found that the time displayed by the version command is the time of the first build, x86 and arm are the same.

> printf("%s\n", buildinfo_date()); 

And you can set the time using `date -s`．

alexkalmuk, 2020-03-10:
> Hi, I found that the time displayed by the version command is the time of the first build, x86 and arm are the same.
> 
> > printf("%s\n", buildinfo_date());
> 
> And you can set the time using `date -s`．

Sure, thanks, it's an invalid issue.

closed 2020-03-10 by alexkalmuk