embox/embox #1690: Building with make 4.3

Deryugin, 2020-02-18:
Building Embox with make 4.3 causes following error:

```
> make
 MKGEN mk_core_def.mk
mk/core/define.mk:1463: *** recipe commences before first target.  Stop.
make[2]: *** [mk/load-mk.inc.mk:165: mk/.cache/mk/mk_core_def.mk] Error 2
make[2]: *** Deleting file 'mk/.cache/mk/mk_core_def.mk'
make[1]: *** [mk/main.mk:30: build] Error 2
make: *** [Makefile:37: all] Error 2
```

Temporary solution is to downgrade `make` to 4.2.

Seems like the problem is caused by this function in `mk/core/define.mk` at line 1379: `__builtin_to_function_inline`.

Deryugin, 2020-02-19:
As it turned out, this error was caused by using `builtin_func-def-ifdef` before it was processed by `$(def_all)` macro. This is fixed in 29d16f7

But now we have a new problem: segmentation fault. This is caused by stack overflow during processing `mk/load-mybuild.inc.mk:95`. I have spend some time figuring out how exactly `builtin_func-` stuff works, but it seems to be a bit messy, so I don't know better solution than just increase stack size. In Linux it's just 8KiB by default. Dirty fix for that is done in 9bb869f.

Any suggestions how to handle this problem more elegant?

Deryugin, 2020-02-20:
Another issue with make 4.3 is building third party modules.

`openlibm` happened to run `build` stage before `patch` while building parallel (i.e. `make -jN`), I suggest to fix by adding explicit dependencies for each building stage e08281e

I wonder why this problem was not found previously. Am I missing something?

alexkalmuk, 2020-02-20:
Hm, what about third_party applications that do not use all these targets, just some of them - like `make download extract patch` only? Lile `@Build(stage=1,script="$(EXTERNAL_MAKE) download extract patch")`

Maybe it would be ok, but should be checked/

alexkalmuk, 2020-02-20:
I see third-party/bsp/stmf4cube/Mybuild use only 3 targets, so if it will be built, so ok.

alexkalmuk, 2020-02-20:
Try to rebuild `arm/stm32f4-discovery` once it built (without clean). You will get an error

Deryugin, 2020-02-20:
Thanks for your comment! You're right, I'll try to fix it now.

Deryugin, 2020-02-20:
Fixed with 42749ef6866f831df94884d58b5b74300cf75636
When files are specified as dependencies (instead of recipes) it doesn't attempt to extract/patch/build it second time.

alexkalmuk, 2020-02-20:
Yes, now it was built fine :)

Deryugin, 2020-02-20:
I think we're ready to merge  #1692 as there are no more known issues?

alexkalmuk, 2020-02-20:
Yes, of course, I approved PR.

PR #1692 (2020-02-21): mk: Add real fix for make 4.3
Closes #1690
closed 2020-02-21 by Deryugin