embox/embox #1696: Wronge backtrace on ARM

anton-bondarev, 2020-02-21:
To reproduce you can add `assert(0;` or simply add a breakpoint in gdb to critical_dispatch() and look at stacktrace

bartekxk, 2020-02-23:
Hello, I would like to work on this issue but, what is exactly wrong with this backtrace?

Thanks.

alexkalmuk, 2020-02-24:
I will try to explain this on another example:

I used the following patch to add `assert(0)` in src/cmds/net/ifconfig.c:

```diff
diff --git a/src/cmds/net/ifconfig.c b/src/cmds/net/ifconfig.c
index dbd9ea0..a2df5c9 100644
--- a/src/cmds/net/ifconfig.c
+++ b/src/cmds/net/ifconfig.c
@@ -285,6 +285,8 @@ int main(int argc, char *argv[]) {
        struct ifconfig_args args;
        void *addr;
 
+       assert(0);
+
        iface = NULL; /* XXX required for -O2 */
        memset(&args, 0, sizeof args);
```

Ok, what is expected behavior of this code? When `ifconfig` command will be executed it will go into `main()` then into `assert(0)`. `assert(0)` expression is _always_ false, so assert will be failed and Embox stack trace should appear on your screen.

Then, what Anton says about in this issue? He says that you can get an _actual_ stack trace from gdb to compare with stack trace generated by Embox:

```
(gdb) b ifconfig.c:main
Breakpoint 1 at 0x23e10: file src/cmds/net/ifconfig.c, line 288.
(gdb) c
Continuing.

Breakpoint 1, main (argc=9, argv=0x9ded050 <__pool_storagethread_pool+261832>)
    at src/cmds/net/ifconfig.c:288
288		assert(0);
(gdb) bt
#0  main (argc=9, argv=0x9ded050 <__pool_storagethread_pool+261832>)
    at src/cmds/net/ifconfig.c:288
#1  0x0009c3c0 in cmd_exec (cmd=0x121dfc <__mod__embox__cmd__net__ifconfig+20>, 
    argc=9, argv=0x9ded050 <__pool_storagethread_pool+261832>)
    at src/framework/cmd/core.c:35
#2  0x0002068c in run_cmd (data=0x1c9ab04) at src/cmds/shell/tish.c:172
#3  0x00009688 in task_trampoline (
    arg_=0x9dcd310 <__pool_storagethread_pool+131464>)
    at src/kernel/task/multi.c:51
#4  0x0000a9e4 in thread_trampoline () at src/kernel/thread/core.c:80
#5  0x0000a944 in thread_self () at src/kernel/thread/core.c:63
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

Here we connected to Embox through gdb and set a breakpoint at ifconfig.c:main() function. You can then go step by step or just continue execution.

If I continue execution I will get in Embox shell:

```
 ASSERTION FAILED on CPU 0
	at src/cmds/net/ifconfig.c:288
	in function main

0
Data abort exception!
regs:
00090078 09deceac e59ff018 fffffff8
00124000 00000000 00000000 00000000
00000000 00000000 00000000 09deceac
0000001f 00002e3c
```

So you can see this stack trace is not actual stack trace like in GDB - it's Data Abort.

You can check what stack trace Embox will generate for another configuration: e.g. `x86/qemu`

Re-build Embox and run. You will get the following stack trace:

```
 --   00001000 * A R    thread 7  task 2 --------------------------------------

  6 0x001450c9 <__assertion_handle_failure+0xf6> ...pat/libc/assert/assert.c:45
  5 0x00121d27 <__module_embox__cmd__net__ping_text_vma+0x0> ...clude/swab.h:46
  4 0x00184204 <cmd_exec+0x50>                      src/framework/cmd/core.c:23
  3 0x0011f10b <run_cmd+0x77>                         src/cmds/shell/tish.c:159
  2 0x00109a8c <task_trampoline+0x1d>                src/kernel/task/multi.c:47
  1 0x0010a87d <thread_trampoline+0x91>             src/kernel/thread/core.c:69

run 0x00109a6f <task_trampoline+0x0>                 src/kernel/task/multi.c:47
```

Ok, so it is much more similar to an actual stack trace then in case of `arm/qemu`.

alexkalmuk, 2020-02-24:
Hm, probably there are several bugs here. We are still discussing this issue. we will let you know soon what to do further.

alexkalmuk, 2020-02-24:
The problem was in the ARM function frame format, so it was solved in #1711. After we added the corresponding compiler flags stack trace became fine.

@bartekxk Sorry, the issue was invalid, you have to choose a new one (we are updating the list)

closed 2020-02-24 by alexkalmuk