embox/embox #1698: Improve 'uname' core util command

anton-bondarev, 2020-02-21:
Now `uname` use mybuild options for getting platform(-i), processor(-p) and systems (-o) use some other way to get it.

Nishchay-sopho, 2020-02-24:
I would like to work on this issue.
I have looked at the uname.c and Uname.my files and basically got an idea about how the workflow of the code is but am not able to understand what exactly do you want from this.
Could you please guide me how to get started?

anton-bondarev, 2020-02-24:
Hi
The simplest thing in this task is get rid of `option string system = "Embox-OS"` from module description. You can just change it to a macro in C code

Then you can add a module with simple analog sysctl. For example you can look at uname().

Nishchay-sopho, 2020-02-24:
So, that means you want the information regarding the processor, platform and system to be modular as the things in struct utsname.
But we still would have to hard code the value of the macros for the system, platform and processor in the file where the macro is, wouldn't we?

anton-bondarev, 2020-02-24:
Yes, I want to remove Embox dependency 
`#include <embox/cmd.h> /* options */`

Of cause,it's better if you do that without hard code at all. I mean in sources of 'uname' command. 
We can look at [https://github.com/wertarbyte/coreutils/blob/master/src/uname.c](https://github.com/wertarbyte/coreutils/blob/master/src/uname.c) and see that processor and platform get by sysctl

Nishchay-sopho, 2020-02-24:
I have done a small implementation to improve this functionality in 17d310b8a51fc149051baa8e82187853757e3547 . 

Could you please let me know if or not this is what was desired ?

anton-bondarev, 2020-02-25:
Yes, it looks like what I wish.
But you should prepare separate PR for this task and wait when it'll merge before start another issue.

Nishchay-sopho, 2020-02-25:
Oh yeah, I didn't realize that while pushing it to the master of my forked repository. 

I will remove this commit for now and wait for #1712 to get merged and then push this one and create a PR for the same.

Is it okay?

anton-bondarev, 2020-02-25:
Yes, it's ok. Thanks.

alexkalmuk, 2020-03-13:
@Nishchay-sopho Hi, how goes? If you are not going to continue with this issue we can unassign you from this issue.

kssuraaj28, 2020-11-27:
Hi, can I try this one?

kssuraaj28, 2020-11-27:
Hi, I would like to try this :)

anton-bondarev, 2020-11-27:
> Hi, I would like to try this :)

@kssuraaj28 Ok, I've assigned it with you

kssuraaj28, 2020-11-28:
I have a question regarding the build systems and Makefiles: In `mk/load-mk.inc.mk`, we have lines like this: `$(mk_core_def) : ALLOC_SCOPE := a` --> What does this mean?
It doesn't seem like a 'normal' make rule to me.

anton-bondarev, 2020-11-28:
> I have a question regarding the build systems and Makefiles: In `mk/load-mk.inc.mk`, we have lines like this: `$(mk_core_def) : ALLOC_SCOPE := a` --> What does this mean?
> It doesn't seem like a 'normal' make rule to me.

Yes, it is not usual syntaxis. Embox has its own languages (DSL) and it has been developed in Make.
In fact, Make is a functional language, and in it you can not only describe the rules, it is just very effective for this aims. And also Make as an ordinary language it is not very good. I find it difficult to answer specifically about this place, but it seems to me that this is a variable formed somewhere and for it (all its members) the rule is described.

But why have you looked at this part?

kssuraaj28, 2020-11-28:
The build system for embox looked interesting and different (I also couldn't find much documentation on it) - I was just looking at the source code for the build system.
Moreover, my after reading the source code, my understanding is that the user-mode/kernel-mode hardware features of processors are **not** used (this is understandable as it is an embedded system meant to run dedicated applications). Is this correct?

anton-bondarev, 2020-11-29:
> The build system for embox looked interesting and different - I was just looking at the source code for the build system.
> Moreover, my after reading the source code, my understanding is that the user-mode/kernel-mode hardware features of processors are **not** used (this is understandable as it is an embedded system meant to run dedicated applications). Is this correct?

Yes, the usual using cases are without user and kernel mode separating.
But Embox is configurable OS and for some systems, it can be used both with isolated address spaces and different privilege levels.  It's not about this issue of course.

anton-bondarev, 2020-12-09:
Closed #2277

anton-bondarev, 2021-02-24:
Closed #2277

closed 2021-02-24 by anton-bondarev