embox/embox #1699: Improve 'lsblk' command

anton-bondarev, 2020-02-21:
Now 'lsblk' print block dev size in bytes, but it's useful to show in (K bytes, M bytes, G bytes)

ahmedkrmn, 2020-02-22:
Hi Anton,
Do you have an idea where I can get started to implement this?

alexkalmuk, 2020-02-22:
Hi Ahmed, you can start with launching this command in Embox shell. And of course, it is not a bad idea to try this command on your host Linux system.

anton-bondarev, 2020-02-22:
Hi,
You can use x86/qemu template with virtual (or real) USB mass storage as described on [wiki](https://github.com/embox/embox/wiki/Using-USB-on-qemu)  and call lsblk in command line.

ahmedkrmn, 2020-02-22:
Thanks @anton-bondarev and @alexkalmuk. I'm working on the implementation now.
I just want to ask if there is a simpler way for testing my changes locally instead of running the following each time I make a change?
```
make confload-x86/qemu
make
./scripts/qemu/auto_qemu 
```

ahmedkrmn, 2020-02-22:
I tried and tested an implementation which works when compiled and ran separately through `gcc ./lsblk.c && ./a.out`.
When I modify the `lsblk.c` file in the `src/` directory and re`make` the project, then I run `lsblk` in the embox shell, the following error message keeps on being printed indefinitely:

```
 --   00000000 * A   W  thread 1106247325  task 4 -----------------------------

 62 0x0014b0f1 <exception_handler+0x113> ...h/x86/kernel/exception_handler.c:20
 61 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 60 0x001021fc <stack_iter_next+0x4c>    src/arch/x86/lib/debug/stack_iter.c:40
 59 0x0010071f <do_traceback_dump_thread+0x4d>     src/lib/debug/whereami.c:135
 58 0x0010088f <whereami+0x88>                     src/lib/debug/whereami.c:170
 57 0x0014b0f1 <exception_handler+0x113> ...h/x86/kernel/exception_handler.c:20
 56 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 55 0x001021fc <stack_iter_next+0x4c>    src/arch/x86/lib/debug/stack_iter.c:40
 54 0x0010071f <do_traceback_dump_thread+0x4d>     src/lib/debug/whereami.c:135
 53 0x0010088f <whereami+0x88>                     src/lib/debug/whereami.c:170
 52 0x0014b0f1 <exception_handler+0x113> ...h/x86/kernel/exception_handler.c:20
 51 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 50 0x001021fc <stack_iter_next+0x4c>    src/arch/x86/lib/debug/stack_iter.c:40
 49 0x0010071f <do_traceback_dump_thread+0x4d>     src/lib/debug/whereami.c:135
 48 0x0010088f <whereami+0x88>                     src/lib/debug/whereami.c:170
 47 0x0014b0f1 <exception_handler+0x113> ...h/x86/kernel/exception_handler.c:20
 46 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 45 0x001021fc <stack_iter_next+0x4c>    src/arch/x86/lib/debug/stack_iter.c:40
 44 0x0010071f <do_traceback_dump_thread+0x4d>     src/lib/debug/whereami.c:135
 43 0x0010088f <whereami+0x88>                     src/lib/debug/whereami.c:170
 42 0x0014b0f1 <exception_handler+0x113> ...h/x86/kernel/exception_handler.c:20
 41 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 40 0x001021fc <stack_iter_next+0x4c>    src/arch/x86/lib/debug/stack_iter.c:40
 39 0x0010071f <do_traceback_dump_thread+0x4d>     src/lib/debug/whereami.c:135
 38 0x0010088f <whereami+0x88>                     src/lib/debug/whereami.c:170
 37 0x0014b0f1 <exception_handler+0x113> ...h/x86/kernel/exception_handler.c:20
 36 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 35 0x001021fc <stack_iter_next+0x4c>    src/arch/x86/lib/debug/stack_iter.c:40
 34 0x0010071f <do_traceback_dump_thread+0x4d>     src/lib/debug/whereami.c:135
 33 0x0010088f <whereami+0x88>                     src/lib/debug/whereami.c:170
 32 0x0014b0f1 <exception_handler+0x113> ...h/x86/kernel/exception_handler.c:20
 31 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 30 0x001021fc <stack_iter_next+0x4c>    src/arch/x86/lib/debug/stack_iter.c:40
 29 0x0010071f <do_traceback_dump_thread+0x4d>     src/lib/debug/whereami.c:135
 28 0x0010088f <whereami+0x88>                     src/lib/debug/whereami.c:170
 27 0x0014b0f1 <exception_handler+0x113> ...h/x86/kernel/exception_handler.c:20
 26 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 25 0x001021fc <stack_iter_next+0x4c>    src/arch/x86/lib/debug/stack_iter.c:40
 24 0x0010071f <do_traceback_dump_thread+0x4d>     src/lib/debug/whereami.c:135
 23 0x0010088f <whereami+0x88>                     src/lib/debug/whereami.c:170
 22 0x0014b0f1 <exception_handler+0x113> ...h/x86/kernel/exception_handler.c:20
 21 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 20 0x001021fc <stack_iter_next+0x4c>    src/arch/x86/lib/debug/stack_iter.c:40
 19 0x0010071f <do_traceback_dump_thread+0x4d>     src/lib/debug/whereami.c:135
 18 0x0010088f <whereami+0x88>                     src/lib/debug/whereami.c:170
 17 0x0014b0f1 <exception_handler+0x113> ...h/x86/kernel/exception_handler.c:20
 16 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 15 0x001021fc <stack_iter_next+0x4c>    src/arch/x86/lib/debug/stack_iter.c:40
 14 0x0010071f <do_traceback_dump_thread+0x4d>     src/lib/debug/whereami.c:135
 13 0x0010088f <whereami+0x88>                     src/lib/debug/whereami.c:170
 12 0x0014b0f1 <exception_handler+0x113> ...h/x86/kernel/exception_handler.c:20
 11 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 10 0x001021fc <stack_iter_next+0x4c>    src/arch/x86/lib/debug/stack_iter.c:40
  9 0x0010071f <do_traceback_dump_thread+0x4d>     src/lib/debug/whereami.c:135
  8 0x0010088f <whereami+0x88>                     src/lib/debug/whereami.c:170
  7 0x0014b0f1 <exception_handler+0x113> ...h/x86/kernel/exception_handler.c:20
  6 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
  5 0x001021fc <stack_iter_next+0x4c>    src/arch/x86/lib/debug/stack_iter.c:40
  4 0x0010071f <do_traceback_dump_thread+0x4d>     src/lib/debug/whereami.c:135
  3 0x0010088f <whereami+0x88>                     src/lib/debug/whereami.c:170
  2 0x0014b0f1 <exception_handler+0x113> ...h/x86/kernel/exception_handler.c:20
  1 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49

run 0x00109a90 <task_trampoline+0x0>                 src/kernel/task/multi.c:47
[... 1 lines cut ...]
[... cut ...]
  2 0x0011f6f3 <system_start+0x79>           src/init/system_start_service.c:27
  1 0x00100958 <kernel_start+0x15>                         src/kernel/init.c:23

run 0x0014e8ff <boot_stub+0x0>                src/kernel/sched/boot_thread.c:20
EXCEPTION [0xe]: error = 00000000
EAX=41effe9d    EBX=00240000 ECX=0000000a EDX=00000000
 GS=00000010     FS=00000010  ES=00000010  DS=00000010
EDI=00000000    ESI=0000000a EBP=099fc8ac EIP=001021fc
 CS=00000008 EFLAGS=00000046 ESP=00000000  SS=00000051

```

ahmedkrmn, 2020-02-22:
@anton-bondarev and @alexkalmuk. The error message occurred when I was trying to display the `float` size using `printf()` and `%f`. I tried to use `sprintf()` with `%f` to convert the `float` size to a `string` but the above error message also occurred. Typecasting the `float` size to `uint64_t` before printing is the only option that worked.
Is there a certain reason for this behavior?

alexkalmuk, 2020-02-23:
> Thanks @anton-bondarev and @alexkalmuk. I'm working on the implementation now.
> I just want to ask if there is a simpler way for testing my changes locally instead of running the following each time I make a change?
> 
> ```
> make confload-x86/qemu
> make
> ./scripts/qemu/auto_qemu 
> ```

Embox is statically configurable (it's one of the project concepts), it means you have no ability to run some pre-compiled binary within Embox. In particular, yes, it means you need to rebuild embox to verify your changes (basically, only your modified file will re-compiled and linked then to final embox image).

alexkalmuk, 2020-02-23:
> @anton-bondarev and @alexkalmuk. The error message occurred when I was trying to display the `float` size using `printf()` and `%f`. I tried to use `sprintf()` with `%f` to convert the `float` size to a `string` but the above error message also occurred. Typecasting the `float` size to `uint64_t` before printing is the only option that worked. Using `round()` also caused this issue.
> 
> Is there a certain reason for this behavior?

It's due to math library (`embox.compat.libc.math_builtins` in your conf/mods.config) floating operations missing implementations. I think you have to switch to `embox.compat.libc.math_openlibm`  and try once more.

ahmedkrmn, 2020-02-23:
Hi can I do that?

alexkalmuk, 2020-02-23:
You can delete `embox.compat.libc.math_builtins` module from your conf/mods.config and add `embox.compat.libc.math_openlibm` instead. It's just to check floating point version will work.

But I think your solution with `uint64_t` probably will be ok because we can use lsblk without floating-point support (hardware and software) on the target platform. Still, we need to think about that.

anton-bondarev, 2020-02-24:
@ahmedkrmn I agree with Alex, using uint64_t looks useful for at this place

PR #1705 (2020-03-19): Mirror Linux lsblk command
Fixes #1699 

- This PR improves the available `lsblk` command in Embox. It mirrors the linux [lsblk](http://man7.org/linux/man-pages/man8/lsblk.8.html) command adding the `-b, --bytes` flags to choose the unit of displaying size.

    [image]

_I tried multiple ways to round the value of size to the nearest integer but they all failed resulting in the error message posted on the issue page. The current behavior is typecasting the `float` to `uint64_t` (flooring)._
closed 2020-03-19 by anton-bondarev