embox/embox #1701: 'blockdev' command

anton-bondarev, 2020-02-22:
Implement of 'blockdev' command ([man 8 blockdev ](http://man7.org/linux/man-pages/man8/blockdev.8.html)
The first implementation may include only `--getbsz` command or `--report` options.

Example, getting of blockdev information may be found in 'lsblk' command

bw-space, 2020-02-22:
I want to take up this issue, can I work on it?

anton-bondarev, 2020-02-22:
yes, you can :)

bw-space, 2020-02-22:
Thanks a lot, I have been going through the source code, I have seen implementations of many shell functions, but I am not able to figure out where are they called from,like:
if I type 'uname', how is it called, or where should I place my new command, so that it is found by the tish shell. Is there documentation of such related queries?

anton-bondarev, 2020-02-23:
You can get information about how to add a new command to Embox on[ "Quick start" wiki page](https://github.com/embox/embox/wiki/Quick-Start#hello-world-application) or in embox_quick_start_en.pdf from [ documentation repo](https://github.com/embox/embox-docs/releases)

bw-space, 2020-02-23:
Thanks, I have managed to make the first app, I am able to get the block size. What all entries should be present in the --report part?
'RO    RA   SSZ   BSZ   StartSec            Size   Device' all from here?

anton-bondarev, 2020-02-23:
Block size and block_dev size are enough now

bw-space, 2020-02-24:
Can you let me know, if the code in PR is alright, or needs some changes?

anton-bondarev, 2020-02-24:
@bw-space Can you prepare PR according to the rules adopted by Embox ?

worldinmyfist, 2020-02-24:
Can I work on this issue as well?

Deryugin, 2020-02-25:
@worldinmyfist no. Please consider some other issue that is not assigned to anyone.
We will add more issues with "Good first issue" tag soon.

anton-bondarev, 2020-06-29:
Closed by #2008

closed 2020-06-29 by anton-bondarev