embox/embox #1702: service_run() does not consider result of starting service

anton-bondarev, 2020-02-22:
```
static int service_run(const char *path, char *const argv[]) {
	pid_t pid;
	pid = vfork();

	if (pid < 0) {
		int err = errno;
		printf("vfork() error(%d): %s", err, strerror(err));
		return -err;
	}

	if (pid == 0) {
		execv(path, argv);
		exit(1);
	}
	service_add(path, pid);

	return 0;
}
```

in spite of execv() can be broken service always adding to services list

It's blocked by #1335

worldinmyfist, 2020-02-24:
Hi can I work on this issue?
I need a starting point if you could help me.

Deryugin, 2020-02-25:
@worldinmyfist sure! Basicly, you should make sure that process is properly created with `execv()` and print something meaningful if something went wrong.

Codestyle and git workflow can be found on our wiki: https://github.com/embox/embox/wiki

worldinmyfist, 2020-02-25:
Hi it would be great if you could tell me in which file is this function located?

Deryugin, 2020-02-25:
@worldinmyfist `service_run()` is in src/cmds/service.c. I advise to use `grep` command to find functions quickly.
```
> grep service_run -rn ./src/
./src/cmds/service.c:92:static int service_run(const char *path, char *const argv[]) {
./src/cmds/service.c:133:	return service_run(command, &argv[1]);
```
`service_run` argument is used to pass the pattern you're looking for, `-r` (recousrive) flag makes it go into directories recoursively, `-n` (number) makes it print line number where pattern was matched

worldinmyfist, 2020-03-06:
Hey 
I went through the code and did not find any possible errors except in execv() function
Thus if issue #1335 is resolved, this issue should be solved.
I want you to assign me other issue so that I can work on that .
Thank you

worldinmyfist, 2020-03-06:
The command kill(parent, KILLSIG) runs when execv() fails. When this happens, service_add() will not get executed, and thus will not be added to the service list. Thus one of the solutions can be execv() returning to previous running environment after failing. Thus if #1335 is solved this issue should be solved.

alexkalmuk, 2020-03-10:
@worldinmyfist Ok, it seems it's actually blocked by #1335, you can choose another issue.

Ahish9009, 2020-03-16:
Hi!
Could I pls get some more info on how to test if this is working properly. I am testing whether my fix to #1335 is working, and thought I'd try it out on this too.

alexkalmuk, 2020-03-16:
@Ahish9009 You can try `service qwerty123`, then `service` and see non-existent qwerty123 is in list.