embox/embox #1709: Extend block_dev_test command with new options

alexkalmuk, 2020-02-24:
* Add parameter which specifies the block number at which the command should start at.
* Add parameter which specifies blocks count to be tested.
* Add option to read/write multiple blocks in a single iteration (currenly, we read/write only single block in one iteration)

ymdatta, 2020-02-24:
Hi, i would like to work on this issue. Can i take it up?
Thanks.

alexkalmuk, 2020-02-24:
@ymdatta Yes, you can. But first of all : I cannot find your proposal in email. Did you send your proposal to anybody of the mentors?

alexkalmuk, 2020-02-24:
We received your email, thanks,  so now I will assign you to this issue.

alexkalmuk, 2020-02-26:
Hi @ymdatta, thank you for the https://github.com/embox/embox/pull/1727.

You probably faced these problems with `-i` and `cd0`, so I think it's better to test block_dev_test new options with another stuff:

```
./scripts/qemu/qemu_with_disk.sh

```
In Embox:
```
root@embox:/#block_dev_test hda                                                 
Starting block device test (iters = 1)...
iter 0...
Testing 0 (of 16384)
Testing 1 (of 16384)
Testing 2 (of 16384)
Testing 3 (of 16384)
...
```

ymdatta, 2020-02-26:
Hi @alexkalmuk, thanks for the suggestion.

When i started trying to solve this issue (#1709), i realized i need something other than `cd0` to test my changes with. So, i followed this [Using-USB-on-qemu](https://github.com/embox/embox/wiki/Using-USB-on-qemu) and right now i am using USB to test my changes.

alexkalmuk, 2020-02-26:
Nice, are you using `ehci_pci` or `ohci_pci` in your mods.config?

ymdatta, 2020-02-26:
I tried using `ochi_pci` it didn't work, but with `ehci_pci` it's working. :)

alexkalmuk, 2020-02-26:
> I tried using ochi_pci it didn't work

Yeah, it's already known bug :) I'm just working on it right now.

alexkalmuk, 2020-02-28:
@ymdatta I merged your contribution, so before to start with the third bullet please rebase on the top of our master.

ymdatta, 2020-02-28:
@alexkalmuk I will be keep that in mind before committing. Thanks.

ymdatta, 2020-03-03:
@alexkalmuk : For third option, do we need to take how many blocks to read/write in a single iteration as a command line option from the user? or do we fix the number of multiple blocks?

Deryugin, 2020-03-04:
@ymdatta I think that best way is to provide option to specify number of multiple block and have some default value so we don't require user to specify it every time. Exact default value doesn't matter, say 4 or 8 or 32 blocks (but make sure it's a sane value which would fit regular block devices).

alexkalmuk, 2020-03-05:
@ymdatta I am agree with @Deryugin, let's add an option.

ymdatta, 2020-03-05:
@Deryugin , @alexkalmuk : Sure, i will do that.

closed 2020-03-09 by alexkalmuk