embox/embox #1713: Implement memaling for heap_simple memory allocator

alexkalmuk, 2020-02-24:
There are several memory allocation algorithms in Embox, one of them is heap_simple - the very basic one, which is rarely used but is useful sometimes for debugging. However, the current heap_simple implementation misses `memalign()` function, so it's required to implement it.

bartekxk, 2020-02-24:
Hello, I would like to work on this issue.

alexkalmuk, 2020-02-24:
Hello, great!

bartekxk, 2020-02-24:
I created pull request. Can you review it?
 #1715 
Thanks.

bartekxk, 2020-02-27:
You can close this task.

alexkalmuk, 2020-02-27:
Yes, of course, thanks. btw you will use something like "Close #<issue>" or "Fix #<issue>" it will be closed automatically - https://help.github.com/en/github/managing-your-work-on-github/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword

alexkalmuk, 2020-02-27:
Closed by https://github.com/embox/embox/pull/1715

closed 2020-02-27 by alexkalmuk