embox/embox #1714: Extend lsusb command

alexkalmuk, 2020-02-24:
Add the following option to lsusb command:

(from `man lsusb`)
```
       -v, --verbose
              Tells lsusb to be verbose and display detailed information about the devices shown.
              This  includes  configuration  descriptors  for  the device's current speed.  Class
              descriptors will be shown, when available, for USB device  classes  including  hub,
              audio, HID, communications, and chipcard.
```

We already have `lsusb` in Embox, but it only prints current devices, it would be great if we can see also device descriptors at least.

This issue requires to study USB, but only very basic high-level things, without going deeper into the USB subsystem. You have to study how to get a USB descriptor and show this descriptor to user in a nice way.

bartekxk, 2020-02-24:
Hello, I would like to work on this issue.

alexkalmuk, 2020-02-25:
Let's finish with https://github.com/embox/embox/issues/1713 first. Then you can to choose a new one :)

bartekxk, 2020-02-25:
Sure, I will also look at other tasks and after finishing my, I will let you know again :).

shivamanipatil, 2020-02-26:
I woould like to work on this one.

anton-bondarev, 2020-02-27:
@shivamanipatil You've been assigned with the issue.
Please look through our code style ( [https://github.com/embox/embox/wiki/Code-Style](https://github.com/embox/embox/wiki/Code-Style)) and git roles ( [https://github.com/embox/embox/wiki/Contributing%3A-Git-cheatsheet](https://github.com/embox/embox/wiki/Contributing%3A-Git-cheatsheet) ) to prepare PR.

shivamanipatil, 2020-02-27:
@anton-bondarev @alexkalmuk 
[image] 
Right now I am priting this is this fine?

alexkalmuk, 2020-02-27:
Hi, the pattern looks good but I would recommend printing Configuration Descriptor instead of Interface Descriptor.

alexkalmuk, 2020-02-27:
But the pattern looks nice :)

shivamanipatil, 2020-02-27:
**struct usb_dev** contains **usb_desc_device** and **usb_desc_interface** and doesn't contain **usb_desc_configuration**. So is it fine if I add **usb_desc_configuration** to **struct usb_dev** for displaying configuration descriptors.

PR #1737 (2020-02-29): cmds/hardware/lsusb: Extending lsusb with -v flag
Closes #1714 with printing device, configuration and interface descriptors for -v flag.
closed 2020-02-29 by alexkalmuk