embox/embox #1716: Add tests for pthread_kill

alexkalmuk, 2020-02-25:
There is POSIX function `pthread_kill` - you can read more about this function in `man 3 pthread_kill`. Please, add tests for this function.

nilCux, 2020-02-26:
Hi, I have sent a mail about participating GSoC from chenglin.theo.xu@gmail.com, may I work on this issue please? If I didn't misunderstand, it concerns adding a test for Embox version of `pthread_kill` which should do exactly the same thing as POSIX `pthread_kill`, right?

anton-bondarev, 2020-02-27:
> it concerns adding a test for Embox version of pthread_kill which should do exactly the same thing as POSIX pthread_kill, right?
Yes pthread_kill the same.
How to add Embox unit test you can see if look at src/test/posix/pthread/

nilCux, 2020-03-08:
Hi, sorry to bother you. I'm afraid I need some help. I just wrote a unit test, however I don't know how to run it. I put it under "embox/src/tests/posix/pthread" and added a module in the"Mybuild" file in this path. Afterwards, what should I do to run it?

nilCux, 2020-03-09:
Hi, I have finished a unit test including three cases:

-  A regular case of sending signal to a running thread
- Send signal 0 to a running thread
- Send signal to a terminated thread

However, there are three things I need to verify:

1. Using pthread_kill to send signal 0 should not be considered as an error, right?
2. Using pthread_kill to send a signal to a terminated thread will cause a crash, what should I do about that?
3. In at least Ubuntu, when I use pthread_kill to send a signal to a terminated thread, the return value of pthread_kill is 3. Is this value POSIX? Should I use this value in my test?

alexkalmuk, 2020-03-09:
All the things you asked are pretty good questions!

> Using pthread_kill to send signal 0 should not be considered as an error, right? Currently, it returns -1 in this case, I think it's wierd. At least in Ubuntu, the return value is 0.

https://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_kill.html says
`As in kill(), if sig is zero, error checking shall be performed but no signal shall actually be sent.`

When we take a look at kill() now - https://pubs.opengroup.org/onlinepubs/9699919799/functions/kill.html we can see `If sig is 0 (the null signal), error checking is performed but no signal is actually sent. The null signal can be used to check the validity of pid.`

So yes, it looks like zero sig can be used to check if the thread is correct and do nothing.

> Using pthread_kill to send a signal to a terminated thread will cause a crash, what should I do about that? Should I add this part of test into comment, so when the problem is fixed, we can test by un-commenting this part of test?
> In at least Ubuntu, when I use pthread_kill to send a signal to a terminated thread, the return value of pthread_kill is 3. Is this value POSIX? Should I use this value in my test?

If embox crashes in case of sending sig to terminated thread it's a bug in embox for sure. You could create the issue for this case. You can push this test case of course, and comment it.

nilCux, 2020-03-09:
Thank you for answering my questions. I just pushed my work. Please let me know if I need to improve or modify it.

PR #1807 (2020-03-16): add test for pthread_kill
Close #1716 Done with several problems.
- Send signal 0 with pthread_kill will return -1
- Send signal to terminated threads behave wierd (Even crashed when the signal is SIGALRM)
closed 2020-03-16 by anton-bondarev