embox/embox #1717: Partitionaly implement 'lshw' command

anton-bondarev, 2020-02-25:
Add 'lshw' command which reporting about hardware in the system. Now show only pci, network and usb devices

layman-n-ish, 2020-02-26:
Hello, I'd like to take this up. Thanks in advance.

alexkalmuk, 2020-02-26:
As far as I can remember you already finished https://github.com/embox/embox/issues/1708, but well, you are free to choose a new one if you want, of course.

layman-n-ish, 2020-02-28:
Hi. In order to print the hierarchy of hardware devices for `lshw`, without the device-tree support, I'm a bit lost. If the hierarchy of the hardware devices is not needed, then, isn't the task just to integrate `lsusb`, `lspci`, etc. in one umbrella utility - `lshw`? Is there any other device configuration I'd have to print in the case of `lshw`? Please do correct me if I'm fundamentally wrong about `lshw`. 

**Note:** Busybox's implementation of `lshw` simply reads and formats off of `/proc` fs.

anton-bondarev, 2020-02-29:
> is the task just to integrate lsusb, lspci, etc. in one umbrella utility - lshw?
Yes!
But if you add also hierarchy of the hardware devices, it's be nice!

anton-bondarev, 2020-02-29:
@layman-n-ish You've already finished # 1708 (one "good first issue") please if you want to take more tasks you should select another (not "good first issue"). Unfortunately, It's difficult for us to create a lot of "good first issue", but we must give an opportunity to solve at least one of them to all who want to take part of GSoC.

Can I unassigned you from this issue?

layman-n-ish, 2020-02-29:
> Unfortunately, It's difficult for us to create a lot of "good first issue", but we must give an opportunity to solve at least one of them to all who want to take part of GSoC.

Apologies, I was not aware of this. 

> May I unassigned you from this issue or you already have some progress?

No, I hadn't made much progress. It's only fair for you to unassign me and give others a chance. Sorry, again, if any discomfort was caused.

anton-bondarev, 2020-02-29:
@layman-n-ish

> Sorry, again, if any discomfort was caused.

no problem, thank you! :)

BK1603, 2020-09-13:
I'd like to take this up!

BK1603, 2020-09-13:
I'd like to work on this :)

BK1603, 2020-09-13:
I'd like to start working on this :)

anton-bondarev, 2020-09-13:
@BK1603 Ok.
I've assigned it with you

BK1603, 2020-09-22:
@anton-bondarev I wasn't expecting this but I have gotten very busy these days. I don't think I'll be able to complete this any time soon. I think it'd be best to unassign me so that anyone else wanting to work on this can take this up.
Sorry for the trouble

anton-bondarev, 2020-09-22:
@BK1603 No problem.

avinal, 2020-12-13:
Hello @anton-bondarev, I would like to work on this issue, actually already working but need some directions 😅 , as you said above in the comments, the first approach would be combining lspci, lsblk and another utility in a single command. That said, my approach is to combine the implementation of all such commands and rewrite them under lshw, am I going in the right direction? because I think it would be mostly duplicate code.  

Another approach could be using already written codes, such that if lshw is called it will internally call lspci and others, with extra information and we could restrict/modify flags to make it proper. Please clear this confusion. Thanks :)

anton-bondarev, 2020-12-14:
@avinal Yes, you are right. The issue can be resolved in different ways. And yes, in the simplest approach (which I described) some code will be duplicated. But It is the "Fist Good Issue" which aims to understand development processes in the project. So I consider, you should solve the issue in the simplest way and then you can improve the implementation (maybe in different issue). For example you can extractduplicate code to a library.

avinal, 2020-12-14:
Cool then, I will do it by simplest approach. You may assign me this issue, I will submit a PR soon :)

anton-bondarev, 2020-12-14:
> Cool then, I will do it by simplest approach. You may assign me this issue, I will submit a PR soon :)

Sure. I've assigned it with you! :)

avinal, 2020-12-15:
Hello @anton-bondarev, 
As discussed, I implemented it by combining related functions from various command implementations. This is the output for my approach. 
[image]

For now, it shows the following infos, (I compared it with ubuntu command output to decide what to include)
- `cpuinfo`
- `lsblk`
- `lsusb`
- `lspci`
- `ifconfig`

Only `-h` flag is supported in this approach. Also, I have added this command to x86/qemu config only.

anton-bondarev, 2020-12-15:
@avinal It looks ok. But you should create PR to discuss more detail

avinal, 2020-12-15:
#2291 Done 👍🏼

PR #2291 (2020-12-23): Add lshw partial implementation
This merge request does the following:-
- Adds  partial `lshw` command implementation
- Closes #1717
closed 2020-12-23 by anton-bondarev