embox/embox #1718: Porting the last version of Openlibm

alexkalmuk, 2020-02-25:
https://openlibm.org/

We already have Openlibm in Embox in `third-party/lib/openlibm/`. It's required to update this version to the most recent one.

alexkalmuk, 2020-02-25:
Hello, thank you, we received your email.

ThePiGrepper, 2020-02-27:
one question regarding the Makefiles, especially regarding external builds. (Im still reading mk/extbld/lib.mk to figure out what exactly the build system does with external builds)
Apparently PKG_SOURCES is the variable defining where to download the sources from, and additional patches can be applied by using the variable PKG_PATCHES. I've been having some issues creating a new patch, exactly like the previous one, but for the new version (commit: 5b0e7e981321687ac0abe711fdeb30adcd9da932 of openlibm), so I was wondering. Would it be acceptable to do all of the patching in a separate git repo(like the one you already have at https://github.com/embox/openlibm), and create the archive from that branch?

alexkalmuk, 2020-02-27:
> Would it be acceptable to do all of the patching in a separate git repo(like the one you already have at https://github.com/embox/openlibm), and create the archive from that branch?

I think it's not a good way because we have many third-party allocations, and we try to download all of them from the external project's resources.

Let's try to figure out how to solve your problems with sources patching.

> Apparently PKG_SOURCES is the variable defining where to download the sources from, and additional patches can be applied by using the variable PKG_PATCHES

Yes, you're right.

> I've been having some issues creating a new patch, exactly like the previous one

You can try `./scripts/app_diff.sh` script to cretate a patch. btw, what's is URL you specified in PKG_SOURCES?

ThePiGrepper, 2020-02-27:
> I think it's not a good way because we have many third-party allocations, and we try to download all of them from the external project's resources.

I see, that's ok then. Let's go with fixing the patching.
 
> You can try `./scripts/app_diff.sh` script to create a patch. btw, what's is URL you specified in PKG_SOURCES?

I changed the URL to a personal server, where I had 'git archived' the latest commit available(w/o patches). I modified the md5sum as well, just to test that it worked. I did this because I didnt know for sure if the current build scheme supported local file paths.
In any case, using app_diff.sh it build successfully.
- I have a couple of questions though. Is there any unit test for checking openlibm(or any other third party software)? If there isn't, I'd still like to test if the library works fine.
- I'd need to upload the new tar.gz to the github archive. You'll probably prefer to do that yourself though. In any case, the changes to the makefile were just a few lines.

alexkalmuk, 2020-02-28:
> I changed the URL to a personal server

Can we try to use https://github.com/JuliaMath/openlibm/archive/v0.7.0.tar.gz as an official source? If you will specify this URL in PKG_SOURCES and md5 (you can calculate it once it is loaded), Embox will automatically save this archive in download/. Then during the build the unpacked sources will appear at `build/extbld/third_party/openlibm` (something like this).

> Is there any unit test for checking openlibm(or any other third party software)? If there isn't, I'd still like to test if the library works fine.

We have `embox.test.math.math_test`, but you have also change the default `include embox.compat.libc.math_builtins` to `include embox.compat.libc.openlibm` in your mods.config.

> I'd need to upload the new tar.gz to the github archive. You'll probably prefer to do that yourself though.

Let's try with https://github.com/JuliaMath/openlibm/archive/v0.7.0.tar.gz

alexkalmuk, 2020-02-28:
I just pointed to the official releases - https://github.com/JuliaMath/openlibm/releases and refer to the latest available one.

ThePiGrepper, 2020-02-28:
ok, done. apparently passes the tests as well on x86/qemu conf. I sent a PR (https://github.com/embox/embox/pull/1742).

PR #1742 (2020-03-02): update openlibm to v0.7.0
Closes #1718, updating openlibm to v0.7.0 with the local patch. Also sets PKG_SOURCES to point to upstream repo.
closed 2020-03-02 by alexkalmuk