embox/embox #1720: sshd (dropbear) rejects second connection

alexkalmuk, 2020-02-25:
There is an sshd server in Embox - third-party/dropbear/ (what dropbear is - https://matt.ucc.asn.au/dropbear/dropbear.html). So it is a third party application ported to Embox.

_How to start dropbear?_
* `make confload-x86/qemu`. Then add `include third_party.dropbear.dropbear` in conf/mods.config, also add `"service dropbear -F"` in conf/system_start.inc. Build embox.
* Start Embox
* Connect to sshd from your Linux host:
```
$ ssh root@10.0.2.16
root@10.0.2.16's password: 
[5] Jan 01 00:00:58 wtmp_write: problem writing /var/log/wtmp: No such entity
root@embox:/#exit
Connection to 10.0.2.16 closed.
```
_What's the problem?_
Try to connect once more:
```
$ ssh root@10.0.2.16
ssh: connect to host 10.0.2.16 port 22: Connection refused
```

This issue is probably not so easy, but we do not expect fast bug fix from you, but the problem investigation with `gdb`.

zhuanyanchengkong, 2020-02-28:
I try to do this work.

zhuanyanchengkong, 2020-02-29:
I added dropbear to embox and checked the running status,
`root@embox://etc/dropbear#service dropbear`
`Service is already running.`
`[5] Jan 0root@embox://etc/d1 01:00:29 Earropbear#ly exit: Failed to daemonize: Address family not supported by protocol family`
Then I used netstat and found the port of 22 was not open, I think maybe this is the reason can't connect the OS via ssh.
I think the first step should be to open the port that dropbear used, but I don't know how to open a port in embox.

zhuanyanchengkong, 2020-03-01:
I  checked the running status of dropbear and found an error :
> root@embox://etc/dropbear#service dropbear                                      
Starting service: dropbear
Service is already running.
root@embox://etc#[5] Jan 01 00:00:14 Early exit: Failed to daemonize: Address family not supported by protocol family

What does this error mean?

alexkalmuk, 2020-03-01:
I cannot answer your question right now since we run dropbear only with `-F` flag and only from the root directory, so the problem you described probably is some another bug. I would recommend you to start with the reproducing the behavior described in the issue. So we can follow the same the way and try to figure out what's happen.

zhuanyanchengkong, 2020-03-01:
Oh! You reminded me.  I use `service dropbear`  because I found that the operation of Dropbear is abnormal and many commands cannot be used. Now I successfully login using ssh：

> eric@Thinkpad-T480s ~/P/embox-0.4.0> sudo ssh -p 22  root@10.0.2.16                                                                                                      
[sudo] password for eric: 
The authenticity of host '10.0.2.16 (10.0.2.16)' can't be established.
ECDSA key fingerprint is SHA256:NhFfPLTp/88yEc0jtMckNOIZT8EKbS4xsYdG+X2REZM.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '10.0.2.16' (ECDSA) to the list of known hosts.
root@10.0.2.16's password: 
root@embox:/#

Before logging in, I used this command in embox：
> root@embox:/#dropbear -F                                                        
[4] Jan 01 01:18:13 Not backgrounding
[4] Jan 01 01:18:21 Child connection from 10.0.2.10:54494
[4] Jan 01 01:18:28 Password auth succeeded for 'root' from 10.0.2.10:54494

At the same time, dropbear cannot be added to Xinetd
> root@embox:/#dropbear -i                                                        
[4] Jan 01 01:31:25 Early exit: Failed socket address: Socket operation on non-socket
tish error: #1

alexkalmuk, 2020-03-01:
> Now I successfully login using ssh

It's good! So you can try to exit and reconnect to reproduce the bug then.

> At the same time, dropbear cannot be added to Xinetd

Embox is not Linux, so we don't have xinetd. Embox provides only posix compatible layer which makes an application executing in the way similar to Linux. But there may be a number of limitations like with Xinetd.

zhuanyanchengkong, 2020-03-01:
Although I have added `"service dropbear -F"`　in conf / system_start.inc.
But before every time I use ssh service, I still need to use the command of  `dropbear -F` in embox.
Otherwise, I can't connect．

zhuanyanchengkong, 2020-03-01:
> At the same time, dropbear cannot be added to Xinetd
> 
> > root@embox:/#dropbear -i
> > [4] Jan 01 01:31:25 Early exit: Failed socket address: Socket operation on non-socket
> > tish error: #1

I found a more fundamental cause due to the error found in the above operation. There seems to be something wrong with the tish service．So I removed tish  `in conf / system_start.inc`
Then recompile and start to embox, and then found that dropbear successfully appeared in the embox startup service. 

> 	test: running embox.test.critical .. done
runlevel: init level is 2
runlevel: init level is 3
runlevel: init level is 4
Default IO device[ttyS0]
>export PWD=/
>export HOME=/
>netmanager
>service telnetd
Starting service: telnetd
>service httpd
Starting service: httpd
>service dropbear -F
Starting service: dropbear
[4] Jan 01 00:00:00 Not backgrounding

Now you can use ssh service directly on the client．

alexkalmuk, 2020-03-01:
> So I removed tish in conf / system_start.inc

You can also add `service dropbear -F` before `tish`, without removing the last one.

zhuanyanchengkong, 2020-03-01:
> > So I removed tish in conf / system_start.inc
> 
> You can also add `service dropbear -F` before `tish`, without removing the last one.

Indeed so，Thanks!

timtim-ww, 2023-07-15:
Looking inside of dropbear it seems that -F option is more about a convenient way to debug, although might be used in cases, as the current embox case, as the main option.
When _DEBUG_NOFORK_ is defined the main process of dropbear server doesn't fork a new process for the sesion opened. Instead it acts as if it were this child itself and proceeds to managing the new session (that also means you cannot have several sessions simultaneously).
Following the logic of a child process, when session is terminated, **exit(errorcode)** is called. In case of a forked process it just terminates leaving the main process up. But with _DEBUG_NOFORK_ **exit(errorcode)** actually quits the main process.

timtim-ww, 2023-07-25:
#2854 PR adds functionality of multi process session handling (process per session) while the main process stays up listening and handling new connections.
That solves the issue with consequent connections, although there's still some malfunction - after about 6 connections dropbear, it seems, denies authentication. Investigation is ongoing.

timtim-ww, 2023-07-29:
Lastly dropbear wouldn't accept more than 5 unauthorized connections from one address, and it uses pipe close event to notify main process on successful authentication. In embox close doesn't notify a process that waits on select()
At the moment we write to the pipe, so that main process gets notified. Further when the case with close notification will be solved we can use dropbear original notification way.
Autotest ssh passes now. Dropbear is capable of handling several connections simultaneously.

anton-bondarev, 2023-10-24:
Closed by #2866 & #2867

closed 2023-10-24 by anton-bondarev