embox/embox #1723: arping doesn't show microseconds

alexkalmuk, 2020-02-25:
```
root@embox:/#arping 10.0.2.10                                                   
ARPING 10.0.2.10 from 10.0.2.16 eth0
Unicast reply from 10.0.2.10 [aa:bb:cc:dd:ee:ff]  0ms
Unicast reply from 10.0.2.10 [aa:bb:cc:dd:ee:ff]  0ms
```

Expected something like that:

```
root@embox:/#arping 10.0.2.10                                                   
ARPING 10.0.2.10 from 10.0.2.16 eth0
Unicast reply from 10.0.2.10 [aa:bb:cc:dd:ee:ff]  0.285ms
Unicast reply from 10.0.2.10 [aa:bb:cc:dd:ee:ff]  0.343ms
```

eboswell, 2020-02-25:
In the example, are those 0.285ms microseconds or milliseconds?

eboswell, 2020-02-25:
I would like to work on this issue

alexkalmuk, 2020-02-25:
> In the example, are those 0.285ms microseconds or milliseconds?

0.285 milliseconds = 285 microseconds :)

alexkalmuk, 2020-02-25:
There are possible troubles with floating-point arithmetic on CPUs with a really small amount of memory and without hardware FPU, so I would suggest you to use integer arithmetic if it is possible here. But any solution will be great, of course.

alexkalmuk, 2020-03-02:
Fixed by https://github.com/embox/embox/pull/1741

closed 2020-03-02 by alexkalmuk