embox/embox #1729: Implement 'du' command

anton-bondarev, 2020-02-26:


bartekxk, 2020-02-26:
Hello, I would like to work on this issue.

alexkalmuk, 2020-02-26:
Hi, yes you can, of course :)

bartekxk, 2020-02-26:
Should the command work on a single file or recursively?

alexkalmuk, 2020-02-26:
Hm, why recursively? :) You should study a bit more what "regular file" from the issue description mean. For example, you can figure out how to check whether a file is a regular file on your Linux host.

But on the other hand, probably we can use `du` not only for regular files, but it's better to discuss it later.

alexkalmuk, 2020-02-26:
Anyway, you can start with regular files, without directories.

bartekxk, 2020-02-26:
I was thinking about recursive reading du documentation on another unix. Currently I did recursively by default with the option of choosing a file (regular file) / directory but I can change it if you want, for example with the -R parameter. I wrote the code, if you want you can see it here:
https://github.com/bartekxk/embox/commit/72ea9b329fc1caed01ae9d5596ed417742385124

This is almost the whole command but I have one problem. When I refer to st_blocks from the stat structure to check how many blocks file is using, I always get 0. Has reading of this parameter (by the stat function) not been implemented?

alexkalmuk, 2020-02-27:
Your version looks good!

>  When I refer to st_blocks from the stat structure to check how many blocks file is using, I always get 0. Has reading of this parameter (by the stat function) not been implemented?

I don't know if `st_blocks` is implemented correctly... Probably @anton-bondarev will answer.

But for `-b` it's better to refer to `st_size` rather than `st_blocks`.

alexkalmuk, 2020-02-27:
On Linux:
```
$ touch 1.txt
$ echo -n 0123456789 > 1.txt
$ du -b 1.txt 
10	1.txt
```

So it would be impossible to get this 10 bytes value from `st_blocks`.

anton-bondarev, 2020-02-27:
> This is almost the whole command but I have one problem. When I refer to st_blocks from the stat structure to check how many blocks file is using, I always get 0. Has reading of this parameter (by the stat function) not been implemented?

You are right. Unfortunately, filling st_blocks isn't implemented now. We'll create a new task for that soon. Except it, when you work on Embox "initfs" you have no block_dev. It places directly in Embox image (like a mapped file system in flat address space)

bartekxk, 2020-02-27:
> On Linux:
> 
> ```
> $ touch 1.txt
> $ echo -n 0123456789 > 1.txt
> $ du -b 1.txt 
> 10	1.txt
> ```
> 
> So it would be impossible to get this 10 bytes value from `st_blocks`.

From Linux man du:

>       Display   values  are  in  units  of  the  first  available  SIZE  from
>        --block-size, and the DU_BLOCK_SIZE, BLOCK_SIZE and BLOCKSIZE  environ‐
>        ment  variables.   Otherwise,  units  default  to 1024 bytes (or 512 if
>        POSIXLY_CORRECT is set).

So is not the size of the memory blocks? It will be probably 10 * (512 or 1024 B blocks).
I think st_size will return the file size, not the memory size that occupies the disk file (in 512B blocks). If I'm right we have to wait for the st_blocks implementation.

alexkalmuk, 2020-02-27:
And yes and no :) You're right disk usage is not the same as file size. But:

```
       --apparent-size
              print  apparent  sizes,  rather  than disk usage; although the apparent size is usually smaller, it may be larger due to holes in
              ('sparse') files, internal fragmentation, indirect blocks, and the like

       -B, --block-size=SIZE
              scale sizes by SIZE before printing them; e.g., '-BM' prints sizes in units of 1,048,576 bytes; see SIZE format below

       -b, --bytes
              equivalent to '--apparent-size --block-size=1'
```

I think because of `apparent-size` using `st_size` is suitable.

alexkalmuk, 2020-02-27:
btw, `-b` is not a common option for `du` command. For example, there is no such option in Busybox version of `du` at all.

I think we can just return a sum of all files sizes for now, and then we can think about how to do it better.

bartekxk, 2020-02-27:
Okay, so I calculate the size based on sb-> st_size. I also added recursive calculation for folders. Can your review it? Thanks.

alexkalmuk, 2020-02-28:
> By the way, I can't link issue in PR, I probably don't have permission

Hm, may be it's related to why your commits are not linked to your Github account. You can see a grey unauthorized icon here - https://github.com/embox/embox/pull/1736/commits/046ea98c6f31f90e321aa4d519f3892738a1e8ab. Probably because of https://stackoverflow.com/questions/26004587/git-commits-are-not-getting-linked-with-my-github-account

alexkalmuk, 2020-03-02:
Thank you! Your contribution was merged.

closed 2020-03-02 by alexkalmuk