embox/embox #1730: Extending file formats for `file` command

Deryugin, 2020-02-26:
There is a `file` command (src/cmds/fs/file.c), currently it just checks if file is RIFF or text file.

This can be extended for ELF, PNG and GIF formats too, and it should be relatively simple.

Optionally, checking for HTML/XML/SVG file formats may be implemented too, but that's a little bit more complicated as header is not in the fixed position.

VivekKamisetty, 2020-02-26:
Hello, I am interested in taking up this issue

anton-bondarev, 2020-02-26:
Hello, ok. I've assigned it to you

Deryugin, 2020-03-04:
This issue is unassigned now, so feel free to pick this task.

El-Mohamed, 2020-03-05:
tell me more I can't understand

checker5965, 2020-03-06:
@Deryugin @anton-bondarev Hi, I've added functionality for ELF, PNG, and GIF files. Can you assign this issue to me so that I can create a PR?

I can also try to implement HTML/XML/SVG functionality.

alexkalmuk, 2020-03-08:
@checker5965 Hi! As far as I can see you already finished https://github.com/embox/embox/issues/1758. Let's leave this issue for other students :)

We do not require to solve more than one good first issue, but if you want you a free to choose any which is not labeled with "good first issue".

alexkalmuk, 2020-03-08:
By the way, if you already implemented ELF and PNG, you can create PR for these two only, but let's leave other formats for students, please.

checker5965, 2020-03-09:
@alexkalmuk Oh okay, in that case I'll just let it be and pick another issue. Hopefully someone else will take this issue up :)

anton-bondarev, 2020-03-09:
> tell me more I can't understand

@Abnaby
[commented above](https://github.com/embox/embox/issues/1730#issuecomment-596199551)

yog-singh, 2020-03-12:
Hello, I'm interested in solving this issue. Please assign it to me.

anton-bondarev, 2020-03-12:
@yog-singh I've assigned it with you

PR #1829 (2020-03-16): Extended PNG, GIF, ELF, XML format for file command
Fixes #1730 
I've added PNG, GIF, ELF, XML file formats to the file command
closed 2020-03-16 by Deryugin