embox/embox #1739: field st_blocks of stat structure have to been filled by stat()

anton-bondarev, 2020-02-28:
field st_blocks of stat structure is not filled by stat() as noticed in #1729.

if a file belongs to some block_dev stat() should fill it with block_dev information. If there are no any block_dev ("initfs" for example) stat() can consider than block_size = 1 byte.

yogeshwaranL, 2020-03-01:
Hi, I would like to work on this.

anton-bondarev, 2020-03-01:
yes!
I assigned the issue with you!

yogeshwaranL, 2020-03-06:
I see that kfile_fill_stat() gets the file information from the inode. But inode doesn't have the information about block size, also in the definition of stat structure is given st_blocks as no. of 512B blocks. So can I divide the file size by 512 and round up to set the no. of blocks? And how do I get access to block size of a device if I need to get block size from the particular block device?

alexkalmuk, 2020-03-08:
> I see that kfile_fill_stat() gets the file information from the inode. But inode doesn't have the information about block size, also in the definition of stat structure is given st_blocks as no. of 512B blocks. So can I divide the file size by 512 and round up to set the no. of blocks?

May `512` to be Linux specific? When I refer to https://pubs.opengroup.org/onlinepubs/007908775/xsh/sysstat.h.html I see  
```
blkcnt_t  st_blocks  number of blocks allocated for this object
```

> And how do I get access to block size of a device if I need to get block size from the particular block device?

You can find this here - src/drivers/block_dev/

InfernalKn1ght, 2023-09-29:
Hello, can I work on this issue?

niklvdanya, 2023-09-29:
Hi, I would like to get assigned.

anton-bondarev, 2023-09-30:
Hello @InfernalKn1ght , Yes. I've assigned it with you

anton-bondarev, 2024-07-01:
Closed by #2974

closed 2024-07-01 by anton-bondarev