embox/embox #1740: Incorrect output of more command

alexkalmuk, 2020-02-28:
```
root@embox:/#more hosts

*** many newlines goes here ***

127.0.0.1 localhost 
```

These new lines shouldn't appear on the screen.

MrSquanchee, 2020-02-29:
Hii, Alex kalmuk.
I would like to work on this issue.
And may you brief me on how to reproduce the bug and which files are the culprit.

anton-bondarev, 2020-02-29:
@MrSquanchee Ok, I've assigned it with you.

> And may you brief me on how to reproduce the bug and which files are the culprit.
It's described in issue
```
root@embox:/#more hosts

*** many newlines goes here ***

127.0.0.1 localhost 
```
When you start Embox you see command line and you have to create 'more' command which can be called from the command line.
You should start from our [wiki Quick-Start](https://github.com/embox/embox/wiki/Quick-Start)

layman-n-ish, 2020-02-29:
Hi, I had looked into this previously; maybe some of my observations could be useful to y'all. 

I think the `*** many newlines goes here ***` part is just `curses'` attempt to flush the screen (which is of size [80 * 24](https://github.com/embox/embox/blob/master/src/compat/posix/curses/curses.c#L29) - hence we see 24 `\n` chars) when `initscr()` is called - [here](https://github.com/embox/embox/blob/master/src/compat/posix/curses/curses.c#L112), the stdscr's buffer is being initialized with whitespaces and when `doupdate()` is [called](https://github.com/embox/embox/blob/master/src/compat/posix/curses/curses.c#L124) after this, the attempt to create a flushed screen starts which can be seen [here](https://github.com/embox/embox/blob/master/src/compat/posix/curses/curses.c#L246)! So, changing `' '` to `0` in [this](https://github.com/embox/embox/blob/master/src/compat/posix/curses/curses.c#L112) line, i.e. to initialize the buffer with zeros rather than whitespaces (just like the virtual screen's buffer is [initialized](https://github.com/embox/embox/blob/master/src/compat/posix/curses/curses.c#L114)), should remove those multiple newlines although I'm not sure if that is the behaviour required. 

Some useful references I found while reading more about `ncurses`:
[1] [man - newterm](https://linux.die.net/man/3/newterm)
[2] [man - doupdate](https://linux.die.net/man/3/doupdate)
[3] [Introduction to ncurses by Reha K. Gerçeker](https://www.tldp.org/pub/Linux/docs/ldp-archived/linuxfocus/English/Archives/lf-2002_03-0233.pdf)

anton-bondarev, 2020-02-29:
@layman-n-ish You are right, but this task is easier a bit :)

To understand the issue you should run embox and try to do 
`root@embox:/#cat hosts`
Then you should add a command which just output file content.
After previous steps, you may decide how you can implement required features (it's not necessary to implement full functionality in this task) with curses or not.

MrSquanchee, 2020-02-29:
@layman-n-ish Nope that doesn't solve the problem i.e. It doesn't remove the `*** many newlines goes here ***`.
@anton-bondarev I encountered one more problem during this the EOF of is not getting detected.
I tried using feof() but that function is not yet defined.

MrSquanchee, 2020-02-29:
@anton-bondarev . I saw the code for cat.c . Is implementing `more` like `cat` expected ? Doesn't it defeat the purpose ??

anton-bondarev, 2020-02-29:
> @anton-bondarev I encountered one more problem during this the EOF of is not getting detected.
> I tried using feof() but that function is not yet defined.

Thanks, Can you open a new issue or maybe even implement feof()  yourself?

anton-bondarev, 2020-02-29:
> @anton-bondarev . I saw the code for cat.c . Is implementing `more` like `cat` expected ? Doesn't it defeat the purpose ??

I'm not sure now, but I think you can start from it

MrSquanchee, 2020-02-29:
@anton-bondarev . I implemented `more` using your method i.e. not using `curses.c`. and It works.
[image]
[image]

Shall I now proceed implementing it using `curses.c` ??

layman-n-ish, 2020-02-29:
> @layman-n-ish Nope that doesn't solve the problem i.e. It doesn't remove the `*** many newlines goes here ***`.

@MrSquanchee Huh, that's odd. My output after making the aforementioned changes looks something like this: 

[image]

I don't mean to deflect your attention but with this approach, notice the first line (80 chars) of every file viewed is not printed (it is being read though). I wasn't able to get to the bottom of this bug - perhaps cursor positions are being misplaced. Hope you solve the main bug nevertheless. All the best!

MrSquanchee, 2020-03-01:
@anton-bondarev .I solved the problem. Thanks to @layman-n-ish for pointing out the replacement of `' '` with `0`   [here](https://github.com/embox/embox/blob/master/src/compat/posix/curses/curses.c#L112). But that was a partial solution as `endwin()` was [called](https://github.com/embox/embox/blob/master/src/cmds/fs/more.c#L31) before the drawing on the screen. and the Implementation of `endwin()` is still [incomplete](https://github.com/embox/embox/blob/master/src/compat/posix/curses/curses.c#L303) and it doesn't re-maneuver the cursor position.
@anton-bondarev . what's next ... ??

anton-bondarev, 2020-03-01:
@MrSquanchee Ok, maybe. It looks believable. Can you prepare PR for your suggestion?

MrSquanchee, 2020-03-04:
@alexkalmuk . Can we close this issue now ????

alexkalmuk, 2020-03-04:
@MrSquanchee Sure, but if you would link this issues to your PR then it will be closed automatically - https://help.github.com/en/github/managing-your-work-on-github/linking-a-pull-request-to-an-issue

closed 2020-03-04 by alexkalmuk