embox/embox #1743: Poor test suits for posix/time routines

anton-bondarev, 2020-02-29:
There are some tests for asctime/mktime/strftime/time routines locate in 'src/test/posix/time' folder but they is very poor. Test for time() for example check only 'time(NULL)' case.

Extend test coverage to suggest some more test cases for the routines.

sk2sat, 2020-02-29:
Can I work on this issue?

anton-bondarev, 2020-02-29:
@sk2sat You can, but ...

You've already finished # 1722. Thanks, it's good work. 
Please if you want to take more tasks you should select another (not "good first issue"). Unfortunately, It's rather difficult for us to create a lot of "good first issue", but we must give an opportunity to solve at least one of them to all who wants to take part of GSoC.

Shall I assign you with this issue? :)

sk2sat, 2020-02-29:
understood. I will select another task.

KlyukinSA, 2023-07-28:
mktime_test seems to be incorrect. on this data test falls. last line should be `test_assert(t == 994204801);`