embox/embox #1745: math floating point tests

alexkalmuk, 2020-02-29:
There are no basic math tests from FP functions in Embox. Let's add something like `src/tests/math/math_test_fp.c` and test for the following functions:

```
extern double modf(double x, double *i_ptr);
extern double fmod(double x, double y);
extern int signbit(double x);
extern double fabs(double x);
extern float fabsf(float x);
extern long double fabsl(long double x);
```

(it's first several functions from src/compat/libc/math/math_builtins.h)

It's assumed the tests should be very simple, just to make sure the corresponding functions are not stubs and really doing something.

rushilkvs, 2020-02-29:
I would like to work on this issue

anton-bondarev, 2020-02-29:
Thanks. I've assigned it with you.

commit 5f3fb0c9 (2020-03-09): Merge pull request #1772 from rushilkvs/test_for_fp

fixes #1745 math floating point tests
closed 2020-03-09 by alexkalmuk