embox/embox #1747: Add tests for ptread_rwlock_

anton-bondarev, 2020-02-29:
Add unit tests for routines which has been already implemented
```
extern int   pthread_rwlock_destroy(pthread_rwlock_t *);
extern int   pthread_rwlock_init(pthread_rwlock_t *, const pthread_rwlockattr_t *);
extern int   pthread_rwlock_rdlock(pthread_rwlock_t *);
extern int   pthread_rwlock_tryrdlock(pthread_rwlock_t *);
extern int   pthread_rwlock_trywrlock(pthread_rwlock_t *);
extern int   pthread_rwlock_unlock(pthread_rwlock_t *);
extern int   pthread_rwlock_wrlock(pthread_rwlock_t *);
```

puranjaymohan, 2020-03-01:
Hi!
I want to work on this issue.
Please help me with a few basics about the test system, so it becomes easy for me.
Thanks

alexkalmuk, 2020-03-01:
Hi,

> Please help me with a few basics about the test system, so it becomes easy for me.

You can take a look at src/tests, there are many examples here. More specifically, since proposed `pthread_rwlock_*` are posix functions, you can take a look at `src/tests/posix/`.

puranjaymohan, 2020-03-02:
Hi! 
I was trying to build the test and while running the make command I get the following error while linking "undefined reference to `pthread_rwlock_init'"
also, I think the PTHREAD_RWLOCK_INITIALIZER macro is also not implemented.
Please help me with solving this issue.
Thanks

puranjaymohan, 2020-03-03:
Hi I tried writing the tests for these APIs, but when I try the following line of code 

`pthread_rwlock_t rw;`
`test_assert_zero(pthread_rwlock_init(&rw, NULL));`
`test_assert_zero(pthread_rwlock_rdlock(&rw));`
`test_assert_equal(-EBUSY, pthread_rwlock_trywrlock(&rw));`

I get the following error while testing using qemu.

`test: running embox.test.posix.pthread.pthread_rwlock 
	failure at src/tests/posix/pthread/pthread_rwlock_test.c : 18, in function __test_case_at_line_14
		test_assert_equal(-EBUSY, pthread_rwlock_trywrlock(&rw))
	   case at src/tests/posix/pthread/pthread_rwlock_test.c : 14
		"Initialize rwlock with pthread_rwlock_init()"
testing pthread_rwlock (posix/pthread_rwlock_api) failed
		1/1 failures
test: Command returned with code 1: Unknown error code.`

Please help me, it might be a basic error, but I am unable to rectify it being a beginner.
Thanks

anton-bondarev, 2020-03-03:
Hi, I think You should try gdb for debugging.
It's described in README [https://github.com/embox/embox#debugging](https://github.com/embox/embox#debugging)

puranjaymohan, 2020-03-03:
Hi Anton,
I have found the issue, the function `pthread_rwlock_trywrlock()` and `pthread_rwlock_tryrdlock()` should return `EBUSY` when they try to lock an already locked rwlock, but here these are returning `-ENOSYS`, which means, "function not implemented". So should I leave these functions and write tests for the remaining five functions?

anton-bondarev, 2020-03-09:
Thanks. Closed #1770

closed 2020-03-11 by Deryugin