embox/embox #1748: Add some unit test for semafores

anton-bondarev, 2020-02-29:
Add some unit tests for implemented routines from semafore.h
```
//extern int    sem_close(sem_t *);
extern int    sem_destroy(sem_t *);
extern int    sem_getvalue(sem_t *, int *);
extern int    sem_init(sem_t *, int, unsigned);
//extern sem_t *sem_open(const char *, int, ...);
extern int    sem_post(sem_t *);
extern int    sem_timedwait(sem_t *, const struct timespec *); //this
extern int    sem_trywait(sem_t *); //OK
//extern int    sem_unlink(const char *);
extern int    sem_wait(sem_t *);```

Inokinoki, 2020-02-29:
Could I have a try with this?
Luckily, I've ever written some similar codes in one of my courses which aims at implementing a real-time kernel from scratch :)

Inokinoki, 2020-02-29:
And BTW, isn't it semaphore in general? Or there is another meaning in this context?

alexkalmuk, 2020-03-01:
Hi, you should send an email to the mentors first - https://github.com/embox/embox/wiki/Google-Summer-of-Code-2020#introduction

> And BTW, isn't it semaphore in general? Or there is another meaning in this context?

Sorry, I am not sure what do mean when asking about "semaphores in general", but it's posix semaphores, you can read about all of these functions in Linux `man` for example.

Inokinoki, 2020-03-01:
Thanks for reminding me! I've sent you all an email.

> Hi, you should send an email to the mentors first - https://github.com/embox/embox/wiki/Google-Summer-of-Code-2020#introduction
> 
> > And BTW, isn't it semaphore in general? Or there is another meaning in this context?
> 
> Sorry, I am not sure what do you mean when asking about "semaphores in general", but it's posix semaphores, you can read about all of these functions in Linux `man` for example.

Yes, it was a question about the name semaphore. I was doubting whether there is another meaning because in the description "semafore" is used. (from my point of view, Symphony and Symfony are totally different😄). There is no more question about it if it just means the POSIX semaphore.

Deryugin, 2020-03-02:
@Inokinoki yes, that was a typo. We are talking about POSIX semaphores here.

PR #1800 (2020-03-16): posix: semaphore: Add unit tests for POSIX semaphore
Resolve #1748 Add some unit tests for:

```c
extern int    sem_destroy(sem_t *);
extern int    sem_getvalue(sem_t *, int *);
extern int    sem_init(sem_t *, int, unsigned);
extern int    sem_post(sem_t *);
extern int    sem_timedwait(sem_t *, const struct timespec *);
extern int    sem_trywait(sem_t *);
extern int    sem_wait(sem_t *);
```

Implement
```c
extern int    sem_timedwait(sem_t *, const struct timespec *);
```
closed 2020-03-16 by anton-bondarev