embox/embox #1749: add functionality to tish

ThePiGrepper, 2020-03-01:
I was playing around with tish, and soon realized that it didnt have job control(no biggie), nor piping(I kinda missed that one). So I wondered. would it be useful to add some functionality to tish? not a lot of features of course, I understand keeping it minimal is kinda the point, but maybe a couple of basic features. thoughts?

layman-n-ish, 2020-03-01:
Hello. Yes, even I kind of missed pipelining, IO redirection and the most pivotal signal handlers. It'd be fun to add some of these features. Let's wait for others' opinions too!

anton-bondarev, 2020-03-01:
Thanks, I agree. It would very useful

Deryugin, 2020-03-11:
Just a note: POSIX-compatible shell syntax rules may be found here https://pubs.opengroup.org/onlinepubs/9699919799/utilities/V3_chap02.html

If someone wants to implement just subset of those features, please do it with extensibility in mind, so further extensions would not require rewriting everything from scratch.

Adding pipes, redirection, conditions, functions and loops means implementing interpreter, which probably would result in pretty heavy-weight program.

Keep in mind that another few KiBytes of memory are significant for some platforms, so we would need a way to switch between "minimalist" and "rich" shell.

ThePiGrepper, 2020-03-11:
> Just a note: POSIX-compatible shell syntax rules may be found here https://pubs.opengroup.org/onlinepubs/9699919799/utilities/V3_chap02.html
>
> If someone wants to implement just subset of those [...]
Hmm, indeed. that's why I didnt actually thought about POSIX compatibility when talking about tish, just wanted to open a discussion about new _minimal_ features. ie: piping, redirecting. For a fully POSIX shell, it would make more sense to: a) pick a third-party minimal POSIX compliant shell and port it(something a bit smaller than dash though), or b) make Tish modular/expandable as you suggested.