embox/embox #1750: Incompatible behavior of kernel sempahore with POSIX semaphore

Inokinoki, 2020-03-02:
Up to now, in Embox, the POSIX-compliance API of semaphore are implemented in `src/compat/posix/semaphore.c`:

```c
sem_init(sem_t *sem, int pshared, unsigned int value);
sem_destroy(sem_t *sem);
sem_getvalue(sem_t *restrict sem, int *restrict sval);
sem_post(sem_t *sem);
sem_trywait(sem_t *sem);
sem_wait(sem_t *sem);
```

These functions call the semaphore implementations in the Embox kernel (`src/kernel/thread/sync/semaphore.c`). 

The kernel semaphore module works well at the moment:
- `semaphore_leave()` do a release on a semaphore, and decrement `value` field in a `sem_t` structure.
- `semaphore_enter()` do an occupation, and increment `value` field in a `sem_t` structure. 
In this case, while `value` field of a `sem_t` structure is 0, it means the semaphore is totally free. Meanwhile, the case when `value` field equals `max_value` field, means the semaphore is totally busy.

But its behaviors are not compatible with POSIX semaphore.

# POSIX semaphore

According to Linux manual, some of POSIX semaphore API are defined as below:

sem_init() initializes the unnamed semaphore at the address pointed to by sem. The value argument specifies the initial value for the semaphore.

sem_post() increments (unlocks) the semaphore pointed to by sem.

sem_wait() decrements (locks) the semaphore pointed to by sem.

In conclusion, when value of a semaphore is 0, the semaphore is totally occupied. It seems that there is no max value limit in the specification.

# Embox POSIX semaphore implementation

[POSIX-compliance Implementations](https://github.com/Inokinoki/embox/blob/master/src/compat/posix/semaphore.c).
[In-kernel Implementations](https://github.com/Inokinoki/embox/blob/master/src/kernel/thread/sync/semaphore.c).

```c
void semaphore_init(struct sem *s, int val) {
	waitq_init(&s->wq);
	s->value = 0;
	s->max_value = val;
}
```

`sem_init()` calls `semaphore_init()`, which initializes `max_value` field of sem instance to the given initial value, and `value` to 0. *Incompatible*

```c
int sem_post(sem_t *sem) {
	semaphore_leave(sem);

	return ENOERR;
}
```

`sem_post()` calls `semaphore_leave()`, which decrements `value` field of sem_t instance. *Incompatible*

```c
int sem_wait(sem_t *sem) {
	semaphore_enter(sem);

	return ENOERR;
}
```

`sem_wait()` calls `semaphore_enter()`, which increments `value` field of sem_t instance. *Incompatible*

The other functions which call Embox kernel semaphore implementations have the same problem.

Inokinoki, 2020-03-02:
A possible solution is to make an adaptive layer between the two implementations.

Deryugin, 2020-03-02:
@Inokinoki thank you for detailed description!

You're making a valid point.

As far as I can see, the only problem here is the actual `value` and other than that everything is fine (i.e. if you call `sem_init(s, N)` then this semaphore can be locked N times and `sem_post()`/`sem_wait()` work as expected).

I think it can be fixed just by changing `sem_getvalue()` (are there any other ways to tell the state of semaphore?) so it returns `s->max_value - s->value` instead of `s->value`, this way `sem_getvalue()` will return zero for occupied semaphore and value passed to `sem_init()` (i.e. `max_value`) for free semaphore.

Am I missing something?

Inokinoki, 2020-03-02:
Yes, I did it in a small patch: https://gist.github.com/Inokinoki/da97fde29985355eca799ec5be1afd3b

It works well with `sem_wait` (value++)!

But the `value` field may be less than 0, because there is no detection of value 0 in the calling chain `sem_post()`->`sempahore_leave()`. Otherwise, are there some lines which can prevent from it?

```c
void semaphore_leave(struct sem *s) {
	assert(s);
	assert(!critical_inside(__CRITICAL_HARDER(CRITICAL_SCHED_LOCK)));

	sched_lock();
	{
		s->value--;
		waitq_wakeup_all(&s->wq);
	}
	sched_unlock();
}
```

The code is in [semaphore.c Line85](https://github.com/Inokinoki/embox/blob/984621866abdd6914235ba4ca26f8cf572d8ce7b/src/kernel/thread/sync/semaphore.c#L85).

In this case, the `max_value` field can no longer limit the `value` field. Regardless of this, I think what you propose is already a fair solution :smiley: 

I haven't seen yet whether POSIX asks to make a value limitation for a semaphore or not. if POSIX doesn't require it, this could be a good.

But we may need to mention it as comments because some developers may be confused by the two different ways to use.

alexkalmuk, 2020-03-05:
@Inokinoki Thank you for this temporary fix!

Let's discuss also whether we should change the embox's internal semaphores due to POSIX. I think it would be more expected behavior.

PR #1773 (2020-03-05): posix: semaphore: Add an adaptive layer to get POSIX semaphore value
POSIX semaphore value begins from its max available value, `sem_post()` decrements it and `sem_wait()` decrements it. The kernel semaphore module does in contrary.

Temporally resolve #1750