embox/embox #1751: Extend thread command

alexkalmuk, 2020-03-02:
`thread` command shows information about threads:

```
root@embox:/#thread -s                                                          

   id  tid priority   state       time
    0    1      127       W         0s
    1    1        0     R          39s
   12    2      127       W         0s
   13    3      127       W         0s
   14    4      127 * A R           0s
```

Let's add two additional fields - `Task name` and `Stack size`. Expected output:

```
root@embox:/#thread -s                                                          

   id  Tid/Task name priority   state     time Stack size
    0    1/kernel      127       W         0s      8192
    1    1/kernel       0     R           39s      8192
   12    2/telnetd     127       W         0s      8192
   13    3/httpd       127       W         0s      8192
   14    4/thread      127 * A R           0s      8192

```

shivamanipatil, 2020-03-03:
Hey, I would like to work on this.

alexkalmuk, 2020-03-03:
@shivamanipatil As far as I can see you already completed https://github.com/embox/embox/issues/1714. So can I ask to choose issues without the "good first issue" label? (I think new issues will be created soon).

ameetsaahu, 2020-03-05:
Hey, can you assign this to me.
I would like to work on this.

alexkalmuk, 2020-03-05:
Yes, I assigned you.

ameetsaahu, 2020-03-05:
Hey @alexkalmuk I have added Task name, but I am confused about how to get this Stack size done.
Please help me with this.

ameetsaahu, 2020-03-05:
Should I make use of pthread.h?

alexkalmuk, 2020-03-06:
> Hey @alexkalmuk I have added Task name, but I am confused about how to get this Stack size done.
> Please help me with this.

You can try to use `thread_stack_get_size`

Deryugin, 2020-03-11:
Done with #1803

closed 2020-03-11 by Deryugin