embox/embox #1752: ls -la /dev/zero failure

alexkalmuk, 2020-03-02:
Template: `x86/qemu`

```
root@embox:/#ls -ls /dev/zero                                                   
EXCEPTION [0xe]: error = 00000000
EAX=00000004    EBX=00256210 ECX=00000004 EDX=00000004
 GS=00000010     FS=00000010  ES=00000010  DS=00000010
EDI=00000000    ESI=09d08380 EBP=09d081c8 EIP=00151e0f
 CS=00000008 EFLAGS=00000202 ESP=09d081d8  SS=09d0847c

 --   00001000 * A R    thread 15  task 4 -------------------------------------

 12 0x00154a3c <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
 11 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 10 0x00151e0f <strncmp+0x3b>               src/compat/libc/string/strncmp.c:11
  9 0x0016c487 <fs_perm_lookup+0x12a>                   src/fs/syslib/perm.c:58
  8 0x0016b184 <klstat+0x23>                          src/fs/syslib/kfsop.c:293
  7 0x0019d8ff <__module_embox__compat__posix__fs__stat_oldfs_text_vma+0x21> 
  6 0x0019d92b <stat+0x1e>             src/compat/posix/fs/stat/stat_oldfs.c:19
  5 0x001335e8 <main_embox__cmd__fs(long,+0xf4>                        short):0
  4 0x001963de <cmd_exec+0x5c>                      src/framework/cmd/core.c:23
  3 0x0012259c <run_cmd+0x83>                         src/cmds/shell/tish.c:159
  2 0x0010afee <task_trampoline+0x27>                src/kernel/task/multi.c:47
  1 0x0010c002 <thread_trampoline+0xa8>             src/kernel/thread/core.c:69

run 0x0010afc7 <task_trampoline+0x0>                 src/kernel/task/multi.c:47
```

p.s.
It works for example for /dev/ttyS0:
```
root@embox:/#ls -la /dev/ttyS0                                                  
crw-rw-rw-       root       root          0 /dev/ttyS0
```

Dawody, 2020-03-06:
Hi,  
This issue seems interesting
Can you please assign this issue to me?

alexkalmuk, 2020-03-08:
@Dawody Ah, sorry for the delayed reply. Yes, I assigned you.

Dawody, 2020-03-08:
Than you,
- The problem is  this function _get_cdev_tab()_ returns a single element (first element) of devtab struct in _src/drivers/char/char_dev.c_
```
struct dev_module **get_cdev_tab(void) {
        return &devtab[0];
}

```
- We can not Iterate on this single element devfs_lookup() in _src/fs/driver/devfs/devfs_oldfs.c_
```
	for (i = 0; i < MAX_BDEV_QUANTITY; i++) {
		if (0 == strncmp(bdevs[i]->name, name, sizeof(bdevs[i]->name))) {

```
- I fixed the issue by adding a condition to the for loop to be like 
`	for (i = 0; i < MAX_BDEV_QUANTITY && bdevs[i]; i++) {
`
- This may be not the best solution for this issue according to that get_cdev_tab(void) function exactly should do and depending on other functions which use this function.
so I fixed the issue from consumer side not provider side.

alexkalmuk, 2020-03-09:
@Dawody Thank you for your PR https://github.com/embox/embox/pull/1802, I left some comments there.

> The problem is this function get_cdev_tab() returns a single element (first element) of devtab struct in src/drivers/char/char_dev.c
> ```
> struct dev_module **get_cdev_tab(void) {
>        return &devtab[0];
>}
>```

It's not a single element, it's a _pointer_ to the first element of `devtab` array.

Dawody, 2020-03-09:
@alexkalmuk Yes I know, I mean that this is a single element (first element) of devtab array.
- I thought that reaching Null when iterating on the array means that there is no more block devs.
so I added the condition to make it break when reaching Null pointer.
- But I you said that we need to skip when reaching Null so I will modify and push again.
Thank you

Dawody, 2020-03-10:
#1802

alexkalmuk, 2020-03-11:
Closed via https://github.com/embox/embox/pull/1802

closed 2020-03-11 by alexkalmuk