embox/embox #1755: Multiple arguments for `ls` command

Deryugin, 2020-03-02:
Looks like `ls` uses only first non-option argument, so it should be fixed to accept more files as arguments.

Now it works like this:
```
root@embox:/#ls -l shadow
-rw-------       root       root        398 shadow
root@embox:/#ls -l passwd                                                       
-rw-r--r--       root       root        335 passwd
root@embox:/#ls -l shadow passwd
-rw-------       root       root        398 shadow
```

How is it expected to work:
```
root@embox:/#ls -l shadow passwd
-rw-------       root       root        398 shadow
-rw-r--r--       root       root        335 passwd
```

This can be tested on `x86/qemu` or `arm/qemu` templates.

pdzhu, 2020-03-03:
I have solved it，but I don't think my code is very elegant. Later, I'll PR.

anton-bondarev, 2020-03-03:
> I have solved it，but I don't think my code is very elegant. Later, I'll PR.
Ok

PR #1763 (2020-03-03): command `ls` multiple arguments #1755
command `ls` multiple arguments fix
Fixes #1755 

[image]
closed 2020-03-03 by alexkalmuk